bigtreetech/BIGTREETECH-SKR-E3-Turbo #3: Hardware compatibility

CKoewing, 2020-11-11:
Hi,

which additional hardware is compatible?

#1 Can I use my original Ender 3 LCD?
#2 Is it compatible to the [BTT TFT35 E3 V3.0 Touch Screen](https://www.aliexpress.com/item/4000365519309.html)?
#3 Is it compatible to the [BTT DCDC5V V1.0 Module](https://www.aliexpress.com/item/4000474224890.html)?
#4 Is it compatible to the [BTT UPS 24V V1.0 Module](https://www.aliexpress.com/item/4000358794405.html)?
#5 Is it compatible to the [BTT WIFI Module](https://www.aliexpress.com/item/33044587899.html)?
#6 Is it compatible to the [BTT Relais V1.2 Module](https://de.aliexpress.com/item/32972603577.html)?
#7 Is it compatible to the [BTT SFS V 1,0 Smart Filament Sensor](https://de.aliexpress.com/item/4000269547406.html)?

Thanks & best regards,
Chris

chilman408, 2020-11-12:
I have the board coming today but I would assume you need to enable Creality Stock Display to get Item 1 going.  Will report back.

doctorthompson, 2020-11-30:
#2 seems to work just fine.

CKoewing, 2021-07-16:
It is unbelievable that BigTreeTech has not replied to this day. :-(
@bigtreetech : Don't you want to sell your additional products???