bigtreetech/BIGTREETECH-SKR-E3-Turbo #5: Warning, does not fit in Ender 5 electronics case without removing power supply.

lococnc, 2020-11-14:
This board is larger than the other SKR Mini E3 versions and interferes with the power supply in the Ender 5 electronics case.

earth4s, 2020-11-29:
Another fit issue with the Ender 5 Plus. The board hits the mounting poles for the touch screen.

lpla, 2020-12-06:
I did a simple adapter for the power supply that can use short M8 screws and bolts (glued to this adapter): https://www.thingiverse.com/thing:4676935

That way the PSU gives enough space for the board to be drop-in replace. So all components fit inside the electronics case and are hold properly.

Regarding wiring, electronics case fan should be connected to the FAN2 pins. BLTouch can be connected like the SKR E3 mini v2.0 to the Z-PROBE pins (check naming: https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/blob/master/Hardware/BTT%20SKR%20E3%20Turbo-Pin.pdf). See a SKR E3 mini v2.0 guide: https://linsomniac.gitlab.io/post/2020-06-20-ender3-pro-skr-v2_0-bltouch-upgrade/

And I made configuration files for Ender 5 Pro using v2.0.X Marlin with BLTouch:

[Ender 5 or Pro - SKR E3 Turbo config files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5649275/Ender.5.or.Pro.-.SKR.E3.Turbo.config.files.zip)

To compile, just use these config files and set `default_envs = LPC1769` in platformio.ini file.