bigtreetech/BIGTREETECH-SKR-E3-Turbo #6: Hotend Temperature unstable

lococnc, 2020-11-15:
Upgraded from SKR Mini E3 V2.0
Marlin 2.0.7.1

Will not hold temp stable, fluctuates by nearly 10 degrees.
Tried PID Tune and it makes it MUCH worse. Won't even reach set temp if I try PID Autotune.

I checked the heater cartridge wiring, even clipped ends, put on new ferrules and crimped. No luck.

Have I got the right thermistor type specified? Is this board different than the previous ones?

kinder2020, 2020-12-09:
> Upgraded from SKR Mini E3 V2.0
> Marlin 2.0.7.1
>
> Will not hold temp stable, fluctuates by nearly 10 degrees.
> Tried PID Tune and it makes it MUCH worse. Won't even reach set temp if I try PID [...]

I am dealing the same issue here. I was using Triangle lab v6 clone and temp reading was fluctuates by nearly 10 degrees so I ordered the genuine E3D v6 thermistor. the situation got better but I still see up to 5 degree fluctuate during prints.

Invento3D, 2020-12-12:
Hey there, I recently had the same problem and found/fixed the issue. I hope your have some soldering experience! :D
I wrote down the instructions on how to fix the issue here: https://www.reddit.com/r/BIGTREETECH/comments/kar8u7/why_you_should_not_buy_bigtreetech_boards_why/

kinder2020, 2020-12-12:
> hope you have some soldering experience! :D

Thank you! But I already returned it to Amazon. I will try next time if I
encounter this problem. Happy holiday!

Invento3D, 2020-12-12:
> Thank you! But I already returned it to Amazon. I will try next time if I encounter this problem. Happy holiday!
> […](#)

Did you swap the board for a new E3 Turbo or did you go for a complete different model/manufacturer?

manurafal, 2020-12-13:
I have the same problem, but mine fluctuates + - 2 or 3 degrees.

Invento3D, 2020-12-13:
> I have the same problem, but mine fluctuates + - 2 to 5 degrees. Any solution from bigtreetech?.

No. You have to perform the hardware fix or return the board.

droidgee, 2020-12-15:
> > Thank you! But I already returned it to Amazon. I will try next time if I encounter this problem. Happy holiday!
> > […](#)
> 
> Did you swap the board for a new E3 Turbo or did you go for a complete different model/manufacturer? Would be interesting to find out if all E3 Turbo boards are affected or only some

Knock on wood, mine is still working correctly.

Im kinda sorta wondering if some people have incorrect config values...Im sure some are legit hardware issues, though, since this seems to be a pattern with BTT and E3-sized boards.

kinder2020, 2020-12-15:
> > > Thank you! But I already returned it to Amazon. I will try next time if I encounter this problem. Happy holiday!
> > > […](#)
> > 
> > 
> > Did you swap the board for a new E3 Turbo or did you go for a complete different model/manufacturer? Would be interesting to find out if all E3 Turbo boards are affected or only some
> 
> Knock on wood, mine is still working correctly.
> 
> Im kinda sorta wondering if some people have incorrect config values...Im sure some are legit hardware issues, though, since this seems to be a pattern with BTT and E3-sized boards.

I am using the skr E3 v2 for now, but thinking to get another TH3D EZboard. 

For the config value, I have E3D v6 hotend so I always use #5 for hotend thermistor. I tried the firmware from both BTT github and directly from Marlin 4.7.2 but I don't see any different. not sure if there any hidden value/options.

GDRPrint, 2021-01-02:
I received this board Dec 29, and it's working fine, with no issues on the temp as reported in this thread. Perhaps mine was a revision correction of this temperature stability defect. I built my own firmware using the template from Github and config files supplied in the zip. 10 or so prints and everything working as expected, I have the previous 32 bit board on which the card slot failed.
 Hope this helps, and thanks for all the posts around this problem.

droidgee, 2021-01-02:
Yes, there was supposedly a hardware fix for boards after November 30.  Not sure if Amazon has fixed inventory yet, but Im guessing the overseas sources already do.  Mine is still working but only using PLA so Im not sure what results higher temperatures would bring.

Lafin-Larry, 2021-02-14:
Hello, its my first post... i give up, i've been going round for 4 days & just keeps looping me back to the digital connections. I'm swapping my thermistor to PT100 with e3D amp as I install new SKR-E3...
[image]
hmm, image there?  so i have everything but in Pins*.h they have   #define TEMP_0_PIN   PA2 (SKR MINI) when i try TX0 it wont compile/wrong format...  Cant it just be that easy for once? i was hoping to bypass this unstable temp on TH0..  Any help would be very appreciated..
Thank you