bigtreetech/BIGTREETECH-SKR-E3-Turbo #9: Error Compilir REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

manurafal, 2020-11-19:
I'm trying to compile a new marlin firmware 2.0.7.2 with my TFT35 E3 V3.0 but I'm getting the following error:

[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5566702/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5566703/Configuration_adv.h.txt)

Verbose mode can be enabled via `-v, --verbose` option
In file included from f:\ender 3\skr e3 turbo\marlin-2.0.7.2-skr-e3-turbo\marlin\src\pins\pins.h:425,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
f:\ender 3\skr e3 turbo\marlin-2.0.7.2-skr-e3-turbo\marlin\src\pins\lpc1769/pins_BTT_SKR_E3_TURBO.h:237:6: error: #error "Only ZONESTAR_LCD, MKS_MINI_12864, ENDER2_STOCKDISPLAY, and CR10_STOCKDISPLAY are currently supported on the BTT_SKR_E3_TURBO."
  237 |     #error "Only ZONESTAR_LCD, MKS_MINI_12864, ENDER2_STOCKDISPLAY, and CR10_STOCKDISPLAY are currently supported on the BTT_SKR_E3_TURBO."
      |      ^~~~~
CalledProcessError: Command 'arm-none-eabi-g++.exe -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\users\manu\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript(item, exports="env")
  File "C:\Users\MANU\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\MANU\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "F:\Ender 3\SKR E3 Turbo\Marlin-2.0.7.2-SKR-E3-Turbo\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 286:
    apply_features_config()
  File "F:\Ender 3\SKR E3 Turbo\Marlin-2.0.7.2-SKR-E3-Turbo\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\MANU\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "F:\Ender 3\SKR E3 Turbo\Marlin-2.0.7.2-SKR-E3-Turbo\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 262:
    load_marlin_features()
  File "F:\Ender 3\SKR E3 Turbo\Marlin-2.0.7.2-SKR-E3-Turbo\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 250:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\MANU\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\MANU\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)

sl1pkn07, 2020-11-19:
```
237 | #error "Only ZONESTAR_LCD, MKS_MINI_12864, ENDER2_STOCKDISPLAY, and CR10_STOCKDISPLAY are currently supported on the BTT_SKR_E3_TURBO."
```

thisiskeithb, 2020-11-21:
As the error message states, that LCD is not supported.

Use `CR10_STOCKDISPLAY` and connect to EXP3 on the TFT.

thisiskeithb, 2020-11-21:
> I need to compile a Marlin with that option (Reprap_Discount_full_graphic_smart_controller) to be able to use the Marlin screen mode. 

No you don’t.

The E3 Turbo only has a single EXP connector (and uses a single EXP cable), so you can’t connect a `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` because uses two EXP connectors/cables.

Using `CR10_STOCKDISPLAY` will still let you enable “Marlin Mode”.

manurafal, 2020-12-05:
Ok, thanks to all.

closed 2020-12-05 by manurafal