bigtreetech/BIGTREETECH-SKR-E3-Turbo #10: No schematic image

droidgee, 2020-11-22:
No schematic image has been given in the repo.  Is it possible to get one?  It would help with wiring questions (such as: should the 2nd set of DC IN be populated directly from a PSU)

droidgee, 2021-01-13:
Considering all of their other boards have schematics posted.

And there are GPL licensing requirements for the firmware, is it the same for the hardware too?  Im not sure.

asarmelin, 2021-03-07:
Maybe this picture can help you...

[image]

GadgetAngel, 2021-03-07:
> Maybe this picture can help you...
> 
> [image]

The schematic is not the same as the PIN diagram, the schematic shows how the connections are made thru all resistors and capacitors and connectors and other components on the motherboard. For example, if I wanted to remove a pull up resistor I can figure out which one is connected to the Thermistor port I am interested in.

devbotas, 2021-03-28:
Has just been uploaded:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/blob/master/Hardware/BTT%20SKR%20E3%20Turbo-SCH.pdf