bigtreetech/BIGTREETECH-SKR-E3-Turbo #11: Where is the schematic?

casterle, 2020-11-25:
I was able to download the schematic for my SKR Mini E3 V2, and assumed I'd find one for this board as well. Where is it?

TotallyWhack, 2020-12-11:
You're in the wrong repository. This is for the SKR E3 Turbo, not the MINI

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0

GadgetAngel, 2020-12-26:
I agree where is the schematic for the SKR E3 Turbo board?

GDRPrint, 2021-01-01:
Same question, where is the schematic download at BTT ? I see a pic on this video https://www.youtube.com/watch?v=4xtsJ-3-c_U at the 12 min mark, if you want to screen grab.

GadgetAngel, 2021-01-01:
> Same question, where is the schematic download at BTT ? I see a pic on this video https://www.youtube.com/watch?v=4xtsJ-3-c_U at the 12 min mark, if you want to screen grab.

I only see the PINOUT diagram not the schematic diagram

GadgetAngel, 2021-01-13:
Here is Bigtreetech's answer to the schematic diagram:

[image]

devbotas, 2021-03-28:
Has just been uploaded:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/blob/master/Hardware/BTT%20SKR%20E3%20Turbo-SCH.pdf