bigtreetech/BIGTREETECH-SKR-E3-Turbo #13: What is the secret to getting BLtouch to work?

okkman, 2020-12-06:
I have tried every Pin and wiring combo I can think of, pin moves during self test but will not move on command.

radirla, 2020-12-11:
Same issue here! I emailed support with no response yet.

Thinkersbluff, 2020-12-14:
In case this helps: https://www.youtube.com/watch?v=4xtsJ-3-c_U 
Around 40:20, he shows how to wire the BLTouch connectors (may involve swapping some pins in the connectors) and shows where to plug it in.  If that does not do it for you, you may need to get into Marlin and set up things like Force 5V mode, per the ANCTLABS guidance.

These are my settings in Configuration.h:
#define BLTOUCH
#define NOZZLE_TO_PROBE_OFFSET {-47, -10, 0 } // yours will likely differ.
#define PROBING_MARGIN 30  // yours may differ.
#define XY_PROBE_SPEED (133*60)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2) // I had problems at first, so I slowed everything down 
#define MULTIPLE_PROBING 2 //& take multiple samples per point
#define EXTRA_PROBING    1
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     5 // Z Clearance between multiple probes
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20
#define Z_MIN_PROBE_REPEATABILITY_TEST
#define AUTO_BED_LEVELING_BILINEAR
#define RESTORE_LEVELING_AFTER_G28
#define Z_SAFE_HOMING#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // because I use my probe as the Z_MIN_ENDSTOP

These are my Configuration_adv.h settings, for the BLTouch:
  #define BLTOUCH_DELAY 500
 #define BLTOUCH_FORCE_SW_MODE
 #define BLTOUCH_SET_5V_MODE

 NOTE: I did follow these instructions, the first time.  This may be what you need to do, too:
  /**
   * Safety: Activate if connecting a probe with an unknown voltage mode.
   * V3.0: Set a probe into mode selected above at Marlin startup. **Required for 5V mode on** 3.0
   * **V3.1: Force a probe with unknown mode into selected mode at Marlin startup ( = Probe EEPROM write )
   * To preserve the life of the probe, use this once then turn it off and re-flash.**
   */
  //#define BLTOUCH_FORCE_MODE_SET

My BLTouch is working fine, with the Turbo board.  My previous board was the E3 Mini v1.2, with which the BLTouch was also working fine.

Good luck!

scottjvincent, 2021-06-03:
I just follow the previous bltouch thread about it locking up and tried those settings, now it works perfect fixed my BL touch with the new motherboard
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/24