bigtreetech/BIGTREETECH-SKR-E3-Turbo #16: Configuration Data not saving in EEPROM

SenthilKumar8248, 2020-12-27:
I have flashed new marlin firmware 2.0.7.2 with SKR E3 Turbo which is issued by Bigtreetech. If I set new temperature value in "Preheat PLA config" and pressed "store setting". And after restart the Ender-3, the Temperature setting show default value. The value which I stored is not retrieving from EEPROM. Please advise.

Xtreembass, 2020-12-27:
Hi.
I have the same issure, when i usesed a self compiled FW.
When:
Send: M500
Recv: echo:Error writing to EEPROM!
Recv: Error:Field esteppers mismatch.
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Field esteppers mismatch.
Recv: echo:Error writing to EEPROM!
When i use the Orginal BIN i can safe to the EEprom
Still 3 Days i try to figure it out.
Maybe someone can help us here
Thanks a lot

SenthilKumar8248, 2020-12-27:
You are absolutely right. I have the same issue. See screen shot. M500 shows Error writing to eeprom. [image]

[image]

Xtreembass, 2020-12-27:
I also actived in the Firmware: #define DEBUG_EEPROM_READWRITE so i can see what is wrong

trengtor, 2020-12-27:
> I also actived in the Firmware: #define DEBUG_EEPROM_READWRITE so i can see what is wrong

[image]

Xtreembass, 2020-12-27:
The setting is not only uncomment something, type it in. 
it will work

SenthilKumar8248, 2020-12-27:
> I also actived in the Firmware: #define DEBUG_EEPROM_READWRITE so i can see what is wrong

Where is the line of code exist. Whether it's in configuration.h or configuration_adv.h

Xtreembass, 2020-12-27:
i put that in Configuration.h in the last line, fill it in there

rickd1994, 2020-12-30:
i have the same problem eprom does not save

Xtreembass, 2020-12-30:
i try the new bugfix version from yesterday on my testboard, and it works, tomorow i try it on my printer

SenthilKumar8248, 2020-12-31:
> i try the new bugfix version from yesterday on my testboard, and it works, tomorow i try it on my printer
> 
> 

Yes! After using bugfix version, now I can able to save in eeprom.

calvinmclean, 2020-12-31:
Where do you find this bugfix version? Is that from the [Marlin repo](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)? I tried with that branch but ran into some issues with configuration. Not sure if I should be looking somewhere else or just try harder to make the new config changes

Xtreembass, 2020-12-31:
Yes the bugfix from the marlin Homepage, but you have to reload the config from the examples (also on the Homepage).
The bugfix config is differnt to the "normal" Version

calvinmclean, 2020-12-31:
This worked for me as well. Thanks for the help

rickd1994, 2021-01-02:
the bugfix version also works for me thx for the help

SenthilKumar8248, 2021-01-02:
Hi, after using bugfix version, I have new issue "Heating failed: E1 printer halted please restart". I have completed PID Autotune and saved PID value in eeprom, but still getting this issue when I preheat the extruded.
![Uploading 103457803...]

Xtreembass, 2021-01-02:
Detto on my side, also my steppers run too hot, try to lower the Current, now i am at 630 and i run test number : i dont know ;)

SenthilKumar8248, 2021-01-03:
> Detto on my side, also my steppers run too hot, try to lower the Current, now i am at 630 and i run test number : i dont know ;)

Follow the steps shown in this video. Now I have no Heating issue.

https://youtu.be/ZwosxcGmT_Y

dk83, 2021-01-23:
I have the same mistake. No data is saved in the EEPROM.
If the bugfix version is used, "#define EEPROM_AUTO_INIT" is active. 
Then no error message appears on the display - but no values ​​are saved either.

In spite of this, the terminal reports that the EEPROM cannot be initialized.

**Please fix the problem.**

The only interim solution:
Open the file pins_BTT_SKR_E3_TURBO.h.
Then deactivate I2C EEPROM and activate VIRTUAL_SD_EEPROM.
E.g.:
// Onboard I2C EEPROM
// # define I2C_EEPROM
// # define MARLIN_EEPROM_SIZE 0x1000 // 4KB (AT24C32)

// Use Virtual SDCARD eeprom
#define VIRTUAL_SD_EEPROM 8

Xtreembass, 2021-01-23:
The problem is fixed in the last bugfix version

baikal, 2021-01-24:
There is a simple fix if one does not want to use the latest bugfix-2.0.x from Marlin, but e.g 'Marlin-2.0.7.2-SKR-E3-Turbo' from this repo:
Remove or comment line ~51 from **Marlin/src/HAL/LPC1768/HAL.h** which defines a bogus I2C_ADDRESS macro
`
//#define I2C_ADDRESS(A) uint8_t((A) << 1)
`
Then, the i2c EEPROM works just fine.
I guess that what bigtreetech must have done to make the .bin in this repo, as they do work.

dk83, 2021-02-01:
//#define I2C_ADDRESS(A) uint8_t((A) << 1)

That works perfectly for me