bigtreetech/BIGTREETECH-SKR-E3-Turbo #19: When "START PRINT", the M600 (filament change) was kicked in.

BCTANBC, 2021-01-04:
SKR MINI E3 TURBO with TFT35. Latest marlin 2.0.X Bugfix. Using E0-STOP port for Filament runout detection. Old original CR10S filament sensor.

Hi there,
I was having problem with Filament Runout Detection. 
When "START PRINT", the M600 (filament change) was kicked in. The TFT35 popped up with "Filament Runout". This was happened when "FILAMENT RUNOUT DETECTION" was set to either "ON" or "SMART" in TFT35. 
When there is no filament in the sensor, then it will print as usual.

Marlin firmware as shown below: 
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
#define FIL_RUNOUT_ENABLED_DEFAULT true 
#define NUM_RUNOUT_SENSORS 1         
#define FIL_RUNOUT_STATE     LOW      
#define FIL_RUNOUT_PULLUP

*Even after setting RUNOUT_STATE to HIGH, it will just shut OFF the Filament runout sensor function.

Tezzatron81, 2021-07-01:
Im having the same issue but in marlin mode

BCTANBC, 2021-07-01:
FIRST OF ALL, DONT USE FILAMENT SENSOR ON TFT35. ONLY USE THE E3 TURBO BOARD FOR FILAMENT SENSOR.

Tezzatron81, 2021-07-01:
Mine is connected on the e0 plug on the board,in touchmode the sensor doesn't even react, in marlin mode every print states there is a filament change. However i have only tried through octoprint even though this worked on my original cr10s board.

ivan5rov, 2021-07-01:
I am having this problem too - latest bugfix 2.0.x with TFT35 E3. BTT Smart Filament Sensor. I have uncommented "#define FILAMENT_MOTION_SENSOR" aside of the usual settings stated above. Same sensor worked on the original Creality board that came with the printer

dsvi, 2022-07-09:
I don't use TFT35, just SKR E3 v1.2 and i have connected it to PT-DET, just not to bother with rewiring. But i had the same issue. 
Here is my work around.
 
Disable the sensor by deafault in `Configuration.h`

`#define FIL_RUNOUT_ENABLED_DEFAULT false `

And at the begining of print job (after the wait for E and bed temp) i've added this

```
; Enable (and reset) filament runout detection
M412 S1 
```

dsvi, 2022-07-10:
I don't use TFT35, just SKR E3 v1.2 and i have connected it to PT-DET, just not to bother with rewiring. But i had the same issue.
The problem was value of `FILAMENT_RUNOUT_DISTANCE_MM` was too small. But when you change it in firmware it deosn't have any effect, since was already saved on the flash, and read from there. So script like 

```
; Enable (and reset) filament runout detection and set runout distance to 30mm
M412 S1 D30 
M500 ; save settings to flash
```
Helped me.