bigtreetech/BIGTREETECH-SKR-E3-Turbo #22: Marlin firmware config for BigTreeTech SKR E3 Turbo on stock Creality CR-10 and stock CR-10S

Hedda, 2021-01-14:
Anyone tested the new "**SKR E3 Turbo**" board from BTT/Biqu on stock Creality CR-10 or stock Creality CR-10S 3D-printer? 

https://www.biqu.equipment/collections/hot-deals/products/bigtreetech-skr-e3-turbo-32-bit-control-board-integrated-tmc2209-uart-for-ender-3

https://www.bigtree-tech.com/products/btt-skr-e3-turbo.html

https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo

https://github.com/MarlinFirmware/Configurations/pull/223

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%201.4%20Turbo%20TMC2209

https://github.com/MarlinFirmware/Marlin/pull/19500

https://github.com/MarlinFirmware/Marlin/pull/19513

# Description

BigTreeTech SKR E3 Turbo is the next board in the SKR E3 series, but with a LPC1769 MCU and a slightly larger footprint for dual extruders/hotends or independent dual Z depending on firmware configuration.

## Board details:

- 120MHz LPC1769 MCU
- 5 x integrated TMC2209s (TMC2209 s) with Trinamic's Low Power Standby feature* and sensorless homing
- Onboard EEPROM
- 4-layer PCB
- USB port ESD protection via USBLC6-2 diode
- Onboard dual z connectors when using a single Z driver or independent dual Z when repurposing the second extruder driver
- Neopixel support (without needing to mod Marlin/change to third-party libraries)
- TFT support
- Dedicated probe port
- Powerloss detection via external module
- Dual, independently controlled, fan headers
- Power supply control via external module
- Dual extruder filament runout detection
- Onboard thermistor (currently disabled as Marlin doesn't have a "board sensor" feature)

* There will be a separate PR to support Trinamic's Low Power Standby feature, but there are some essential changes in this PR to a couple core Marlin files in the interim, otherwise users will get a TMC CONNECTION ERROR on boot.

I've also put in a PR for the base config (single extruder setup), 2-into-1, and dual Z in the Marlin Configurations repo. Link below.

# Configurations

See the matching config PR in the Marlin Configurations repo: MarlinFirmware/Configurations#223

unrulyrotors, 2021-01-14:
no but I'd almost pay someone to compile a configuration of this for the CR-10 for me. I cannot get it to work for anything.

Amilleriiw, 2021-05-01:
ok so i have  bigtreetech skr e3 turbo  with the bigtreetech tft43 v3.0 it has marlin 2.0 bugfix dual color none mixing geeetech A10 frame i used ender 3 / skr e3 turbo config files from bugfix 2.0 config files marlin and it is up and running very new to this so i had a bunch of problems to work through ,,,, so it prints one color great and when i try to print 2 colors something some where is changing the temperature at color change the set temp starts at 200 then changes to 175 then it stops printing waits for temp 175 then changes temp 190 waits for 190 then changes to 200 waits for 200 then changes color and prints this happens at every color change i have done pid auto tune  so if anyone can give me an idia where to look to fix this it would be very much appreciated

tocguy, 2021-05-02:
Based on only what you've described I would double-check your slicer settings for the two materials you're using. My guess is that one of them has the correct temperature setting and the other material profile does not. Thus, the gcode keeps telling the printer to change temperature when the material changes but the material profile in your slicer does not contain the correct temperature. So, the dumb machine is only doing exactly what it's being instructed to do.

Amilleriiw, 2021-05-02:
Thank you for your response this has been resolved the problem yes lied in the slicer cura turns the hot and off upon tool change switch splicer to prusa and it has the setting to change that and is now working perfectly need to figure out how to change that in cura