bigtreetech/BIGTREETECH-SKR-E3-Turbo #24: bltouch locks system

spunky9296, 2021-01-18:
New board flashed it with stock firmware first and then with the new bltouch zhoming firmware. when i click on bed leveling the system locks up and i cant do anything with the ABL. Any suggestions would be greatly appreciated

dk83, 2021-01-23:
I work with BL-Touch V3.0 on Probe Connection.
This Configuration work on **2.0.7.2-bugfix** Version

//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
   #define ENDSTOPPULLDOWN_ZMIN_PROBE
...
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_ENDSTOP_INVERTING false 
...
#define USE_PROBE_FOR_Z_HOMING
...
#define BLTOUCH

**Extra ToDo for other, like 2.0.7.2 Version**
_You must define pin for probe connection, SKR E3 Turbo use pin P1_22_
Configuration.h:
#define Z_MIN_PROBE_PIN P1_22

scottjvincent, 2021-06-03:
Just tried those settings on mine with bltouch not working and it fixed it, thank you very much