bigtreetech/BIGTREETECH-SKR-E3-Turbo #25: I can't get the neopixel working on the SKR E3 Turbo

lscalpati, 2021-01-18:
I have not been able to find any documentation on how to get this working but I have found and used a number of 1.3 and 1.4 tutorials and have followed them for marlin configuration.  I have been able to install light menu(shows up and is usable) and I have enabled the power on self test.  I have also ordered and installed the DC/DC board and switched the 5v jumper to the opposite side from its default config (middle pin and pin closest to center of board.  Has anyone gotten this working and have any pointers?

OldMogy, 2021-01-31:
Have a look at my configuration files, hopefully they can help you out.
https://github.com/OldMogy/BTT_E3_Ender3Pro_Configs. Note: I did not change the 5V jumper for my configuration as I used an external 5V regulator to power my Neopixel strip.

scottmichaelbarnes, 2021-03-25:
I finally got mine working after some frustration as well.   I have the E3 Turbo board, the old one that has problems, new one on the way.   I followed the latest video on YouTube but I think in that video the guy goes to far.   All I did to get mine working was to start from scratch downloading the E3 Turbo config files from BTT and then modified those and compiled them.   I turned on led menu and neopixels.   One thing I did change was the default pin line in the config.h file.   Commented it out as it by default uses the BTT pin assigned for this.   I think it was pin 2_4.

Two video's to watch are by 3dMakeit and 3D Printscape.

Hope this helps.