bigtreetech/BIGTREETECH-SKR-E3-Turbo #26: Schematic required for Open Source hardware SKR E3 Turbo

gwideman, 2021-01-26:
The SKR E3 Turbo board carries the Open Source hardware logo:

[image]

Use of this logo requires that the vendor provide documentation as follows:

[https://www.oshwa.org/definition/](https://www.oshwa.org/definition/)

"The hardware must be released with documentation including design files, and must allow modification and distribution of the design files. [...] The documentation must include design files in the preferred format for making changes, for example the native file format of a CAD program."

As yet BTT has not posted the schematic diagram. Not only is this required for use of the Open Source hardware logo, the diagram is also needed in order for users to properly identify which MCU signals perform which functions, and thus to be able to adapt firmware accordingly.  

When will we be seeing these schematics?

At this very moment I am in the process of returning (to Amazon) one of these boards because it has the issue with the temperature inputs instability, and lacking a schematic, I cannot verify the validity of the suggested fix on reddit, nor would I be able to address any resulting problems, along with the lack of ability to determine firmware settings that rely on knowing the hardware connections.

A company is obviously  welcome to produce a proprietary board if it is described as such, but part of the value of BTT boards has been that they have schematics available. So it's reasonable that users might expect schematics for new BTT boards, and that expectation is reinforced by use of the Open Source Hardware logo.

devbotas, 2021-03-28:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/blob/master/Hardware/BTT%20SKR%20E3%20Turbo-SCH.pdf

devbotas, 2021-03-28:
Has just been uploaded:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/blob/master/Hardware/BTT%20SKR%20E3%20Turbo-SCH.pdf