bigtreetech/BIGTREETECH-SKR-E3-Turbo #27: E0 will not drive stepper extruder

SpectrumTech, 2021-01-27:
I built a CR10-S5 around this board. I am able to get everything working besides the E0 driver will not move the extruder stepper. This is with cold extrusion protection either disabled or with the hotend at temperature. It will not drive via manual commands from the old menu, from the touch screen or from octoprint. When I switch the wiring from E0 to another driver to test the extruder stepper, it moves fine, but the other axis that has been moved to E0 will no longer move. I believe there is a fault on the board itself between the driver chip and the E0 connector. I have inspected the E0 connector and there are no bent pins or obvious defects.

I am using Marlin Bugfix-2.x firmware and I am using E1 to drive my printers second Z stepper for Z auto align, which is working fine. X,Y, and Z all function, BL touch functions, smart filament detection works - everything but E0.

I have attached my config and pin files if anyone wants to look or see if it works on their hardware.

[CR10-S5_CONFIG.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5882517/CR10-S5_CONFIG.zip)

SpectrumTech, 2021-01-28:
BIGTREE_TECH is not being helpful at all on this. They keep wanting me to troubleshoot the problem myself. Now they want me to send them a video of the problem - which is exactly as I describe. So I really don't get the point. With the size of this printer and the way the enclosure for the electronics is made it is hard to shoot a video while handling a camera on top of changing settings, not to mention dangerous because of all the mains power exposed in the box. so I am just going to return the board while the return window is open and be done with BIGTREE_TECH.

stsweeney41, 2021-04-13:
I agree, there is something wrong between this board and the marlin software. I am going to try  some other firmware as a test and see if that solves the problem.

sgsrtcameron, 2021-05-30:
I am having the same issue right now, did you ever find a solution?

stsweeney41, 2021-05-30:
Unfortunately no... may have to go with a board that has Plug in drivers so
that it is just standalone.

sgsrtcameron, 2021-05-31:
I got mine to work, I will get my code pushed up to my git in the morning, but I'm going to go pass out now. It will be for the ender 5 plus, but it should get you into the ball park.

stsweeney41, 2021-05-31:
Awesome, thanks!

sgsrtcameron, 2021-05-31:
@stsweeney41 https://github.com/sgsrtcameron/Marlin/tree/myconfig/E3-SKR-E3-Turbo-Dual-Z-BLTouch This is for the ender 5 plus, but there should only be a few minor changes for the CR10

sgsrtcameron, 2021-10-18:
@stsweeney41 Was this able to help you at all, Im in the prosses of clossing down this account and didnt want to remove the repo with that commit if you still needed it but had not forked it yet.

SpectrumTech, 2022-02-16:
I got tired of fighting it and abandoned the CR10S5 for a while and then life got in the way. I just now getting back to it and it has the same issues as before, with the latest marlin stable and bugfix release. 

The odd thing is that I upgraded my ender 3 to the BTT SKR E3 Turbo board soon after abandoning the issues with the CR10S5 and it works just fine. 

The two printers are nearly identical except the size of course, but I added a second Z stepper to the Ender 3, all the same accessories and everything works exactly as it should. I even have tried the ender 3 firmware as is on the CR10S5 as is. The only glaring differences at this point is the CR10S5 has a 12V supply and the Ender 3 is 24V and the Y stepper is larger on the CR10S5. I have a silicone bed heater driven by an SSR on the CR10S5, so the bed current draw shouldn't be a factor. I also completely rewired all the stepper harnesses with heavier gauge wire and new connectors just to make sure something wasn't amiss there.

My best guess at this point is that this board, combined with Marlin and a 12V supply is causing the issues. I do have the SpreadCycle chopper timing set to CHOPPER_TIMING CHOPPER_DEFAULT_12V, since that's the only setting I know of in marlin that would be related. I put a DMM on the supply output and it is rock solid while everything that will work is on and in motion, so I don't think its dirty power. I guess I could put an O'scope on it and check for noise/ripple.  I think at this point I'll go through it and swap everything to 24V, which should be the power supply, a few fans and the extruder heater cartridge. Aside from trying to reverse engineer marlin and how it handles stepper drivers, it's about all there is left to try!

closed 2024-04-19 by SpectrumTech