bigtreetech/BIGTREETECH-SKR-E3-Turbo #29: Creality CR-10 or Ender 3 with TMC2209 sensorless homing + BLTouch + Auto Load Filament

Hedda, 2021-02-05:
Has anyone tested SKR-E3-Turbo in a Creality CR-10 or Ender 3 with TMC2209 sensorless homing + BLTouch + original display?

Thinking that could take this configurations example and use as a base for Ender-3, Ender-5, Ender-5 Pro, and CR-10S printers:

https://github.com/MarlinFirmware/Configurations/pull/407

Then mix and merge with these other configuration examples to create new Marlin Firmware config:

- https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo
- https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5/BigTreeTech%20SKR%20E3%20Turbo
- https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5/BigTreeTech%20SKR%20E3%20Turbo%20with%20BLTouch
- https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5%20Pro/BigTreeTech%20SKR%20E3%20Turbo
- https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5%20Pro/BigTreeTech%20SKR%20E3%20Turbo%20with%20BLTouch
- https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%201.4%20Turbo%20TMC2209

badbod, 2021-06-18:
Tested and working, 
Ender 3 V2  with Creality CR-10 12864 ( Ender 3 / CR-10 original display ) - Check 
BLTouch - Check
Sensorless homing - Check
Dual Z using E1 as Z2 stepper driver. - Bonus Check
Dual parallel 20mm pancake motors with BMG extruders on E0 driver using LIN_ADVANCE - Double Bonus Check  :-)

Auto bed leveling with UBL 10x10 Auto Mesh Wizard, Auto Z stepper align, Z-offset Wizard. Babystepping, Hybrid Stealthchop,  etc etc , the worx.
  All features fully tested and functional under Marlin 2.0.9
Also tested with BTT TFT35-E3-V3.0  display in both touch and 12864 (CR-10) modes without issue,

I imagine it would be more than fine for your needs.

Tezzatron81, 2021-07-01:
Badbod mind sharing your source code. Im having issues compiling for my cr10s

badbod, 2021-07-03:
@Tezzatron81 hi, configs attached
[configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/6758609/configs.zip)

badbod, 2021-07-03:
if your still having issues, send a copy of your configs.  Might be easier to see whats happening.

Tezzatron81, 2021-07-04:
badbod that config says its the ramps board not the skr e3 turbo?

badbod, 2021-07-04:
Configuration.h 

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_E3_TURBO
#endif

Tezzatron81, 2021-07-05:
Lol, it happens.

Tezzatron81, 2021-07-05:
badbod i tried to compile and its not going well.ill keep working on it but its throwing errors about the pins

badbod, 2021-07-05:
oohh,    there is a  change in pins for making the E3 Trubo thermistor show as CHAMBER,   err set :
  Configuration.h
#define TEMP_SENSOR_CHAMBER 0    

that should solve the pins issue,  sorry again,  no other pins are changed

Tezzatron81, 2021-07-18:
I did just use it as a guide, it was the controller fan pins that was the problem but i have it sorted now, except the "chamber" doesnt show up on the tft only shiws in marlin mode, but its all working,marlin are working on an actual board thermistor so i can put a thermistor in my chamber etc. cant wait for that to happen.

badbod, 2021-07-24:
Marlin has implemented the board thermistor,  yeh, the board thermistor only showed in marlin mode and as chamber. Havn't checked the latest repo updates for the board thermister addition yet. Think it is stil in bugfix branch?  I did see a commit it was added.

pleasanthill17, 2021-08-19:
> I did just use it as a guide, it was the controller fan pins that was the problem but i have it sorted now, except the "chamber" doesnt show up on the tft only shiws in marlin mode, but its all [...]

 
Tezzatron81 Can you please post your config files. I am trying to the same SKR E3 Turbo on CR10S. 
My skr e3 turbo came with Marlin 2.0.5.3 preloaded and all axis were moving.
I have attached Za and Zb to Z1 and Z2 respectively. I am using Bltouch and it is attached to Bl touch slot (all 5 pins in a row).  Z stop not plugged in. 
I tried compiling with marlin bugfix 2.0.9 and upgraded firmware. 
My y axis is not moving at all. It may have something to do with defining CR10s bed (I am assuming..)
Attached is my config and advanced config file.
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/7011169/Configuration_adv.txt)
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/7011170/Configuration.txt)

Tezzatron81, 2021-08-19:
Id be looking at your wiring. Have u tried flashing the default fw from btt git. Try that first if ur y axis still wont move it's obvious wiring. Because bed size shouldn't have anything to do with the y axis moving

pleasanthill17, 2021-08-19:
1. I flashed the e3 turbo firmware from their GitHub and y axis still didn’t move.
2. I moved y axis with z axis cable at the steppers and it moved confirming that all steppers are working.
3. I rewired my creality v2.1 to the printer and all axis move.

My conclusion is that wiring is fine but board is non functional. Ordered new board. 

Tezzatron81 can you please post your config files, so I can replicate for my new board.

TIA !

Tezzatron81, 2021-08-22:
i dont know how to post my files, sorry

Hedda, 2021-08-24:
> i dont know how to post my files, sorry

Easiest would probably be if fork https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo or https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x by selecting "Add file" -> "Create new file" as trying to making changes in a project you don’t have write access to will automatically fork it and then write it to a new branch in your fork. Later you can choose to link to the file and branch in your fork or send in a pull request to the upstream repository.

Hedda, 2021-08-24:
FYI, @FOG-Yamato has submitted PR with BTT SKR E3 turbo configuration for Sovol SV01, which I think is similar to Creality CR-10S:

https://github.com/MarlinFirmware/Configurations/pull/540

> Independent dual Z configuration
> Standard single extruder configuration
> Refactored author name in previous configurations
> 
> 2-in-1 Hotend configuration missing due to inability to test it

pleasanthill17, 2021-08-28:
> FYI, @FOG-Yamato has submitted PR with BTT SKR E3 turbo configuration for Sovol SV01, which I think is similar to Creality CR-10S:
> 
> [MarlinFirmware/Configurations#540](https://github.com/MarlinFirmware/Configurations/pull/540)
> 
> > Independent dual Z configuration
> > Standard single extruder configuration
> > Refactored author name in previous configurations
> > 2-in-1 Hotend configuration missing due to inability to test it

Thanks for this. Will try and report back.