bigtreetech/BIGTREETECH-SKR-E3-Turbo #30: Error Writing Eeprom

manurafal, 2021-02-07:
[image]
[image]

manurafal, 2021-02-07:
[_Statusscreen.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5939235/_Statusscreen.h.txt)
[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5939237/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5939238/Configuration_adv.h.txt)

manurafal, 2021-02-07:
Ender 3 Pro    Skr E3 Turbo   TFT35_E3_V3
Marlin 2.0.7.2

thrutheframe, 2021-02-08:
In VScode, remove or comment line no.51 from Marlin/src/HAL/LPC1768/HAL.h which defines a bogus I2C_ADDRESS macro 

//#define I2C_ADDRESS(A) uint8_t((A) << 1)

Save and compile. It should work.

manurafal, 2021-02-08:
It is works. PERFECT!!!

Thanks a lot.

> 
> 
> In VScode, remove or comment line no.51 from Marlin/src/HAL/LPC1768/HAL.h which defines a bogus I2C_ADDRESS macro
> 
> //#define I2C_ADDRESS(A) uint8_t((A) << 1)
> 
> Save and compile. It should work.

It is works. PERFECT!!!

Thanks a lot.

closed 2021-05-30 by manurafal