bigtreetech/BIGTREETECH-SKR-E3-Turbo #31: Stepper motors noisy and very hot

pozi78, 2021-02-15:
Hi,

I realizaed that randomly while printing, the steppers start to make noise and become very very hot, more than 80C (untouchable).

After lots of reading and tests, i think the problem comes when some electrostatic hits the machine. I replicated the issue by rubbing my feets into the floor (charging my body) and then touching any of the stepppers except Y. At that moment you can hear te printer make a clack noise and steppers starts to make noise and the temp of the motors rise from 35C to more than 80C. I attach some readings form M122 command before and after the process .
After the problem, the printer keeps moving, but the print is ruined switching layer positions and i think size of the part.
I really like this board, but this problem makes it extremely 
[printing_benchy_after_discharge_reading_1.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5979204/printing_benchy_after_discharge_reading_1.txt)
[printing_benchy_after_discharge_reading_2.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5979205/printing_benchy_after_discharge_reading_2.txt)
[printing_benchy_after_discharge_reading_3.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5979206/printing_benchy_after_discharge_reading_3.txt)
[printing_benchy_before_discharge_reading_1.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5979207/printing_benchy_before_discharge_reading_1.txt)
[printing_benchy_before_discharge_reading_2.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5979208/printing_benchy_before_discharge_reading_2.txt)
[printing_benchy_before_discharge_reading_3.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/5979209/printing_benchy_before_discharge_reading_3.txt)

dangerous to use.

gwideman, 2021-02-20:
Very interesting. 
What kind of printer is this?
When you say "any of the stepper motors", do you mean the X, Z and Extruder motors?
Could you assess which of the motors have their cases grounded, and which do not? 
The fix might be as simple as grounding the motor cases. 
If this is a typical "I3" style printer, with Y = moving bed, then it's tempting to see that the Y motor would be bolted to the chassis (hence grounded), while the Z and Extruder motors are probably separated from ground by the rubber carriage wheels if there's no other ground means. But I would have thought that the Z motor(s) would also be mounted to the chassis and thus grounded, so it would be a surprise if they caused trouble.
At any rate, if static is discharged to an ungrounded motor case, that may well couple capacitively to the motor windings and thus back to the stepper driver chip. It could well disrupt the data stored in the driver chip, including the pulses per step (ie: microstepping ratio) throwing off the layer position and size, and also the current setting, assuming that's software programmed, which I believe it is on these boards (as opposed to a current-setting potentiometer like on older drivers).
Let us know more detail of your machine -- an interesting case study.

pozi78, 2021-02-20:
@gwideman 
Thanks for your attention.

> Very interesting.
> What kind of printer is this?

Sorry, my mistake. It's an Ender 3 Stock except for the SKR E3 Turbo board

> When you say "any of the stepper motors", do you mean the X, Z and Extruder motors?

No, just when i touch the X, Z and E. Y Motor does not have the problem.

> Could you assess which of the motors have their cases grounded, and which do not?

Only Y has its case grounded. I triple checked all my grounds. 

> The fix might be as simple as grounding the motor cases.

I already did grounded Z 4 days ago connecting it to the chasis with a wire. Just Z. Since then i have only one issolated isssue (touching the nozzle with the tweezers the Z motor made the "jump" and became noisy and hot but not the other 3) and i made all kind of tests a bunch of times. Most of the time I can feel the ESD in my fingers but the printer keeps working well as if nothing happpened. If I unbolt the wire that link frame with motor everything fails.

> If this is a typical "I3" style printer, with Y = moving bed, then it's tempting to see that the Y motor would be bolted to the chassis (hence grounded), while the X and Extruder motors are probably separated from ground by the rubber carriage wheels if there's no other ground means. But I would have thought that the Z motor(s) would also be mounted to the chassis and thus grounded, so it would be a surprise if they caused trouble.

I also thought the same, but it turns that the paint of the V-Slot isolates the Z motor. And the part that attach the Z motor to the V-Slot is made from plastic. So totally isolated.

> At any rate, if static is discharged to an ungrounded motor case, that may well couple capacitively to the motor windings and thus back to the stepper driver chip. It could well disrupt the data [...]

At this moment it is clear that the drivers reset when the ESD reach them. It is what they do according to the TMC2209 datasheet section 3.7. I dont know if there is somthing that can be done via firmware to avoid this behaviour. I mean. if an ESD just resetted the drivers, restore them to some default that works and don't convert them into lava.

> Let us know more detail of your machine -- an interesting case study.

aerialviews007, 2021-02-21:
I don't know if this is the issue but I had two stepper motors fail moving from the stock creality board to the BTT SKR Mini E3 v1.2.  I now run the Turbo without issues.

It could be that the original motors cannot really handle microstepping.

pozi78, 2021-02-21:
> I don't know if this is the issue but I had two stepper motors fail moving from the stock creality board to the BTT SKR Mini E3 v1.2. I now run the Turbo without issues.
> 
> It could be that the original motors cannot really handle microstepping.

Do you mean that the steppers don't work with the E3 MINI but they do with the Turbo? I don't have the mini to make any tests, but it is good to know somo more information.

aerialviews007, 2021-02-22:
No, apologies for not being clear.  The Y and Z motors overheated and died using 32 bit boards. The replacements work great and never had a problem since.   I went from stock to Mini E3 to Creality 4.2.7 back to Mini E3 and now I’m on the E3 Turbo.

gwideman, 2021-02-22:
@pozi78 

Thanks for your comprehensive reply.

> At this moment it is clear that the drivers reset when the ESD reach them. It is what they do according to the TMC2209 datasheet section 3.7. I dont know if there is somthing that can be done via [...]

OK, from your discussion of your ability to reproduce the failure, I think we've reached a valid diagnosis -- electrostatic discharge (ESD) to the motor housing is causing a pulse on the windings that reaches the TMC2209s and is sufficient to reset them, which leads to them operating with parameters (presumably current) that cause motor overheating.  

Obviously that ESD could occur from an outside source (such as user who has accumulated a charge) touching one of the motors. But ESD could probably also build up due to the parts of the printer moving about, such as the electrically isolated gantry or trolley with insulating tires constantly rolling along on metal rails.  That gradual buildup by itself would not be the disruptive influence, but if it achieves a voltage sufficient to arc, that might be a pulse that couples capacitively to the motor windings and cause the TMC2209 reset etc.

This raises numerous issues:

1. Presumably the first line of defense is to ground the motor bodies, in fact to ground all the otherwise isolated metal components of the machine.
2. Is there a hardware change that would suppress the ESD pulse and prevent it from disrupting the TMC2209?
3. Is it feasible for the firmware to notice that a TMC2209 has in fact reset -- or at least has changed its settings?
4. If the firmware did notice, then is it feasible to take corrective action that makes sense? If it simply re-configures the TMC2209, that might not be useful, as some time will have gone by during which the TMC2209 has probably been commanding moves at the wrong steps-per-mm scale.

The TMC2209 datasheet or application notes  may have something to say on these points, but I haven't looked yet.

pozi78, 2021-02-23:
> No, apologies for not being clear. The Y and Z motors overheated and died using 32 bit boards. The replacements work great and never had a problem since. I went from stock to Mini E3 to Creality 4.2.7 back to Mini E3 and now I’m on the E3 Turbo.

I am quiet sure that this ESD "isue" caused your motors to overheat. I just received the creality 4.2.7 to try to replicate those tests, and see if it has the same problem, but had no time to assemble. What brand and model are your new motors?

aerialviews007, 2021-02-23:
The replacement motors are also creality.  Dirty power then?

pozi78, 2021-02-23:
> @pozi78
>
> Thanks for your comprehensive reply.
>
>
> OK, from your discussion of your ability to reproduce the failure, I think we've reached a valid diagnosis -- electrostatic discharge (ESD) to the [...]

> Obviously that ESD could occur from an outside source (such as user who has accumulated a charge) touching one of the motors. But ESD could probably also build up due to the parts of the printer [...]

Exactly, if the motors are not grounded, and in fact they aren't by default,  the possibilities of random ESD are very very high.

> This raises numerous issues:
> 
> 1. Presumably the first line of defense is to ground the motor bodies, in fact to ground all the otherwise isolated metal components of the machine.
> 2. Is there a hardware change that would suppress the ESD pulse and prevent it from disrupting the TMC2209?

The datasheet at section 3.7 clearly states some ways of protecting from ESD so there most be something that can be done, but saddly i am not an electronic engineer. But I don't think that a component that is so affected by ESD and prone to fail is reliable in any application. As you said in another issue opened, it is imperative from BigTreeTech to realease the schematics under Opens Source Hardware so they could be reviewed. 

> 3. Is it feasible for the firmware to notice that a TMC2209 has in fact reset -- or at least has changed its settings?

If it is feasible, it could be a "fail safe" but not an actual solution

> 4. If the firmware did notice, then is it feasible to take corrective action that makes sense? If it simply re-configures the TMC2209, that might not be useful, as some time will have gone by during which the TMC2209 has probably been commanding moves at the wrong steps-per-mm scale.

As I said, a "fail safe" but not the solution. 

> 
> The TMC2209 datasheet or application notes may have something to say on these points, but I haven't looked yet.

I looked a bit, but most of it I don't understand. If you do read it and find something please keep us informed.
An official statement from @bigtreetech regarding this delicate issue would be awsome

pozi78, 2021-02-23:
> The replacement motors are also creality. Dirty power then?

My understanding of all of this is that it is not dirty power but wrong or inaccurate parameters of the TMC2209 drivers after a reset caused by ESD which leads to higher current delivered to the motor causing those moters to overheat.

pozi78, 2021-03-01:
Lots of people are having this issue. Not a single word from @bigtreetech 
They don't even answer to emails. Meanwhile my printer is unusable with this board

[https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/34#issue-818866638](url)

yourbuddydinec, 2021-03-02:
The same thing has been happening to me and the only way I have been able to fix it is a reboot. So these days I have to reboot my printer before every print.

yourbuddydinec, 2021-03-06:
Printer just melted another mount from the extruder overheating. I hope we get a response soon.

pozi78, 2021-03-06:
Sad to hear that.

I have an answer from @bigtreetech via youtube's channel Teaching Tech comments. 

https://www.youtube.com/watch?v=LxQWCBE6lLk&lc=Ugy8xcQ97zxF7jhbG-N4AaABAg.9KWY2rA4QCw9KXJf8rhELo

I wrote an email like an hour or two ago. It is saturday and I dont know if they are at work, so lets excpect the answer

yourbuddydinec, 2021-03-06:
Looking at alternate boards since I got mine not terribly long ago and amazon may cover it. Wondering if one of the ones with the slotting in drivers would be a good alternative.

skiwall, 2021-03-11:
I want to join the party. I have 2 E3 turbo boards and they both do exactly this. Printing is fine for some time and the steppers are sitting around 34C. then all of a sudden the steppers start to make a strange noise and heat up quickly. The print also starts layer shifting. the only way to reset is a power cycle of the printer. I've tried stock firmware and compiled my own with the same result. I also fitted a Creality 4.2.7 board after the first board E3 turbo board had this issue and it worked fine. I then fitted a new E3 Turbo boards and the problem returned in the first print.
Interestingly BTT tech support told me that my first board was faulty and they refunded me for it, which is why I purchased a second one. The first one came from Amazon and the second I bought directly from their website and shipped from China. I just can't see that 2 boards shipped from different shops at different times could have the same fault unless it was a design flaw??

gwideman, 2021-03-12:
@skiwall :
> Also the 4.2.7 board uses stepper drivers in standalone mode so they are hardware set for Vref etc therefore even if ESD was sent to them they will not change settings.

Yes, I too noticed that difference between Creality 4.2.7 and BTT E3 Turbo, and came to the same conclusion.

> One final point to ask, is your main PSU still attached to the frame of the printer? 
> mine is not as i'm in an enclosure. Thinking about it the PSU metal casing 
> will be earthed and this is bolted to the frame of the printer so may take any ESD away?

The salient factor (I believe) is whether static charge can build up on some moving part of the printer and then discharge to the windings of the stepper motor, thus pulsing the output stage of the driver chip. A 3D printer is a perfect machine for this -- with its rubber belts and rubber wheels on the carriages. It's basically a Van der Graaff generator, unless the moving subassemblies are grounded!

The TMC2209 and TMC2225 datasheets even say as much:
[http://img.ekeic.com/ekeic/9981564407175.pdf](http://img.ekeic.com/ekeic/9981564407175.pdf)
Section 3.6  "Especially plastic housings and belt drive systems tend to cause ESD events of several kV. It is best practice to avoid ESD events by attaching all conductive parts, especially the motors themselves to PCB ground, or to apply electrically conductive plastic part."

(Here "motors themselves" means the metal cases of the motors.)

skiwall, 2021-03-12:
So, I swapped to my original E3 turbo board as my 1.1.5 board had an issue where the printer would suddenly stop for no reason like it had a reset, heaters went off and the coordinates were lost (like it had been power cycled). When I fitted the 4.2.7 board, this reset issue happened again. ESD would explain this issue also so I don't think it's purely a BTT issue it's just the Creality board went into a failsafe error - i.e. the heaters and steppers turned off (I don't think this is by design) but the BTT board continues and starts to overheat the steppers which is dangerous.
Anyway today I'll add Earth lines to all motors and the frame and retry to see what the effect is with temp monitors on the steppers

skiwall, 2021-03-13:
So I've added earth leads to all motors and the frame and wired them to the PSU case (earth), 10 hour print went smoothly with no failures. Motors stayed at a nice 35 degrees C throughout.

gwideman, 2021-03-13:
> So I've added earth leads to all motors and the frame and wired them to the PSU case (earth), 10 hour print went smoothly with no failures. Motors stayed at a nice 35 degrees C throughout.

@skiwall  Thanks for doing that valuable experiment and reporting back!  That greatly reinforces our suspicions about the TMC drivers getting disrupted and probably reset by ESD events. And that those drivers are susceptible when their parameters are set by software commands over serial communications, instead of being set by hardware jumpers and potentiometer.  And of course good to know what the strategy is to fix that problem.

Could you mention what model of printer you have that lacked grounding? That might help illustrate that this is a commonplace issue, not previously widespread until the influx of software-set driver chips.

skiwall, 2021-03-14:
I have an Ender 3 but crucially I moved the PSU off the frame as it's inside an enclosure. The case of the PSU is earthed and is normally bolted to the printer frame and this would normally provide a good earth for the printer frame (not all steppers, but maybe sufficient earth field to prevent an ESD event). However in my case the printer frame was completely isolated an open to such a problem. As I said earlier though, the ESD problem also effected my 4.2.7 board but in a different way, that board would just stop printing and reset and in one case my LCD screen went funny as well but reset after a power cycle. 
I do think that ESD should be given some more thought in these boards but I was the one who removed the PSU so in some way maybe i'm the problem :)
Anyway I have some more printer testing to do to make sure the earth lead fix is good but i'm pretty confident that it's now fixed.

yourbuddydinec, 2021-03-15:
Anyone seen if the star washer thing from [this video](https://www.youtube.com/watch?v=s1Q3aYO8I9k) can help with this? I really don't want more boards fried.

iZz0s, 2021-03-26:
Hi,

i've just got the same problem and even a reboot don't fix this. Steppers are getting really hot and loud. board is just some days old.

gwideman, 2021-03-26:
@iZz0s says:
> i've just got the same problem and even a reboot don't fix this. Steppers are getting really hot and loud. board is just some days old.

Then you might try the solutions discussed above. Basically you want to drain any static charge build-up on the motors that might well accumulate due to the constant operation of the rubber belts and rubber-tired wheels that operate like a mini Van der Graaf generator. 

First determine whether each motor's case lacks a path to the main chassis ground, by doing a resistance measurement from the motor case to the main chassis.  3D printer motors often lack such a path because they are mounted on components that ride on wheels that have insulating rubber tires.

For those motor cases that don't have such a path, you need to arrange one.  You may be able to connect a ground wire to the motor itself, or to the metal component it's mounted on (but check that the motor case is actually electrically connected to that metal component).

You can test this theory by temporarily adding a clip-lead jumper wire from each motor case to main chassis ground, and see if that cures your issue.

skiwall, 2021-03-26:
As an update, I've been printing all day every day since I added the earth wires and everything is still working normally and i've had no further repeats of this issue

iZz0s, 2021-03-26:
> @iZz0s says:
> 
> > i've just got the same problem and even a reboot don't fix this. Steppers are getting really hot and loud. board is just some days old.
> 
> Are you getting _no_ initial period of proper operation? If that's the case, then it sounds like the board is either misconfigured, or defective.
> 
> But if you're getting at least some initial period of proper operation before seeing the symptoms, then you might try the solutions discussed above. Basically you want to drain any static charge build-up on the motors that might well accumulate due to the constant operation of the rubber belts and rubber-tired wheels that operate like a mini Van der Graaf generator.
> 
> First determine whether each motor's case lacks a path to the main chassis ground, by doing a resistance measurement from the motor case to the main chassis. 3D printer motors often lack such a path because they are mounted on components that ride on wheels that have insulating rubber tires.
> 
> For those motor cases that don't have such a path, you need to arrange one. You may be able to connect a ground wire to the motor itself, or to the metal component it's mounted on (but check that the motor case is actually electrically connected to that metal component).
> 
> You can test this theory by temporarily adding a clip-lead jumper wire from each motor case to main chassis ground, and see if that cures your issue.

Previously, i had installed the skr mini e3 2.0 which was working without problem. My own compiled marlin is 95% the same, just changed plattform/board etc.

Motor currents 600, extruder and dual z has 700. My motors were even after hours around 33-37 Celsius.

I was running this for days now on my skr e3 turbo and it was working as it should, motors even after hours never hotter then 40 Celsius and printer was very silent.

Now my printer is way more loud, even my bmg make bad rattling sounds on retract. The steppers getting 50 celsius in under 5 minutes now. But its not only that, the printer doing now sometimes some crazy stuff. Display dont reacts anymore then, yesterday i tried to print a cube and he doubled the steps in y axis so it wasnt anymore a cube 😀

Reboot doesnt help, so i might try to ground the steppers but im thinking maybe im going back to the e3 mini v2.0.

iZz0s, 2021-04-02:
So, im back to e3 mini v2 and no more motor problems. in the past i had a lot of Problems with skr boards, maybe i will try nexttime mks or lerdge.

yourbuddydinec, 2021-05-01:
Has this issue been resolved yet? I noticed the boards were impossible to get on amazon for a little while and wondered if the ones they have now are safer.

iZz0s, 2021-05-03:
nothing new here, as it's not confirmed from btt. im running skr mini e3 v2 without problems.

yourbuddydinec, 2021-05-06:
I've gotten a new board, a new power supply (mean well) and this is still happening. Seriously we need some answers.

gwideman, 2021-05-06:
> I've gotten a new board, a new power supply (mean well) and this is still happening. Seriously we need some answers.

1. Which "this" are you referring to?

2. What kind of printer does your complaint pertain to?

3. What is the status of the ground on each of the motors for the axes that are affected?

> Seriously we need some answers.

And this present github issue has some answers.  Have you read them?  The condensed version is that, to a highish degree of confidence, the motor cases on all axes need to be grounded, whether directly, or as a side effect of the frame component that they are mounted upon (assuming they are in electrical contact with that frame).

So please let us know the status of your printer, relative to that grounding advice.

If you are wondering why it is that this board is susceptible to that issue, it is because it uses driver chips that have various crucial parameters in internal registers that are set by software, unlike other chips that have a few simple options set by external pins that are hardwired with jumpers.  Those registers can be disrupted by static discharges and change to other values (or reset to default values). And a typical 3D printer with its carriages and gantries rolling back and forth at high speed on insulating rollers is a static-generating machine. And that static charge build-up ultimately discharges in the motor to the windings that are connected to the driver chips.

So priority one is to address grounding of the moving parts of your printer's chassis.

yourbuddydinec, 2021-05-07:
> 1. Which "this" are you referring to?
The steppers growing burning hot to the touch on this board. Hasn't happend on a tested SKR E3 Mini 1.2 or 2.0. Only the E3 Turbo.

> 2. What kind of printer does your complaint pertain to?
Ender 3 OG though with an upgraded meanwell power supply.

> 3. What is the status of the ground on each of the motors for the axes that are affected?
The ones affected are connected directly to the frame.

gwideman, 2021-05-07:
> > 1. What is the status of the ground on each of the motors for the axes that are affected?
> 
> The ones affected are connected directly to the frame.

Just to confirm:  On the Ender 3, as I understand it, there are two independently movable pieces of frame on which stepper motors are mounted, and these would be the X-axis rail that rolls up and down the Z rails, and the head carriage that rolls side-to-side on the X rail.  Unless specifically supplied with a wire connecting these two rolling frame elements to the main chassis, they are not grounded. 

So, though the motors may be connected to the frames on which they are mounted, are the movable frames actually grounded to the main chassis? And that chassis itself grounded?

yourbuddydinec, 2021-05-07:
The ones that have been affected are the extruder motor and the z motor
though which I believe are grounded to the frame. The Y and the X haven't
had a problem as of yet.

gwideman, 2021-05-07:
> The ones that have been affected are the extruder motor and the z motor though which I believe are 
> grounded to the frame. The Y and the X haven't had a problem as of yet.

Hmmm, interesting.  Though I've seen many Ender 3's online, I don't think I've ever seen one in the flesh, and now looking at a few pictures and videos, I'm not so convinced that the Z and Extruder motors are grounded.

The Z motor appears to mount by "hanging" from a black rectangular block attached to one of the vertical posts. If that block is plastic, then the motor is not grounded, at least not via its mounting. Even if it's aluminum, anodizing of the block or post could block the conductivity to the main chassis.

The Extruder motor is mounted on the X-rail assembly, so it's on an assembly that's not intrinsically grounded.

(There is some possibility that if the Z motor _is_ actually grounded, then there may be a ground path to the X rail via the lead screw.)

At any rate, it would still be worth checking whether these motors (cases) are grounded to the chassis.  Simply attach some flexible stranded wire from the main chassis to each of these motors.  You could put a lug under the head of one of the motor's screws, or for real quick-and-dirty, get a metal hose clamp (like used around car radiator hose -- the worm-drive variety) and use that as a conductive means to grip the motor body + clip leads.

yourbuddydinec, 2021-05-07:
The extruder on mine is on the top most rail (with the bowden going down instead of right) which is touching the frame unlike the normal x axis extruder mount. I understand what you mean about the z though. Is there anything I can place under it that is conductive to make sure it is touching the frame? I can do the hose clamp since its the most modifiable to take off unlike soldering a wire. I was hoping for something I could place it on to conduct through the bottom.

All of this still ignores the fact that out of the 3 BTT board's I've tried this is the only one to damage things.

gwideman, 2021-05-07:
> The extruder on mine is on the top most rail (with the bowden going down instead of right) which is touching the frame unlike the normal x axis extruder mount. I understand what you mean about the z [...]

I have no special knowledge about mechanically convenient ways to arrange a conductive path from a stepper motor case to the main chassis ground, especially since I don't have an Ender 3 to look at, and it sounds like yours is a little modified, or perhaps there have been more than one model revision from Creality.

> All of this still ignores the fact that out of the 3 BTT board's I've tried this is the only one to damage things.

I guess we're assuming the failing board is SKR E3 Turbo, and you previously mentioned that you tried SKR E3 Mini 1.2 or 2.0 successfully. Indeed these all have TMC2209's set by UART communications by the MCU, and ultimately by commands sent from the firmware.  So you certainly have a valid point that those boards should be comparable to each other.

But... for you, how long does it take for the overheating issue to occur, and how consistently does it occur? Is it possible that your tests with the Minis were not long enough to provoke the problem? Maybe they were on a day with higher humidity (less likelihood of static buildup)?

It's true that lack of grounding for your motors is not 100% certain to be the problem in your situation. However, it is a situation known to be able to cause this problem, so worth addressing anyway.

And it's very easy to test by using a multimeter to check the resistance between the motor cases and the main chassis case, and between the case and the board ground (at the power input ground terminal, for example).  So rather than spend a lot of time wondering about it, you can either test it with a meter and fix it if needed, or just fix it. Then if the symptom recurs, you at least know that grounding is not the problem.

leech420, 2021-08-09:
Yoo hi. Just want to chime in, I've had two of these boards fail from this, no resolve even after grounding the casings and all of the chassis. Same problems word for word.

I can get the print to last about 6 or 7 hours, then failure. My mode of failure is the same, randomly starts overheating, loud steppers etc, but instead of "missing steps" my board seems to enjoy trying to build from failure point at a 45 degree angle and stretch the part.

tome9111991, 2021-08-28:
You can use in config_adv
`define ADAPTIVE_STEP_SMOOTHING`
For more silent.

KerseyFabrications, 2021-08-30:
That's a good idea! I need to add that!

gwideman, 2021-08-30:
@leech420  
 
> My mode of failure is the same, randomly starts overheating, loud steppers etc, 
> but instead of "missing steps" my board seems to enjoy trying to build from
> failure point at a 45 degree angle and stretch the part in Y axis. Looked firmware
> multiple times, havent had any luck.

This again fits the hypothesis that one or more of the stepper chips are getting fully or partially reset, even though you took measures to ground the relevant electrically separate parts.  In that theory, in one or more of the stepper chips, the register that holds the "microsteps resolution" is getting reset, changing the scaling of that axis.  

However, the 45-degree angle is a puzzle.  Is there any chance your machine is a CoreXY model?

leech420, 2021-08-31:
Appreciate the reply.

I didnt mention my printer is an Ender 5 plus, so yeah core XY. The scaling
theory... is odd. Because its not doubling steps and making the part
physically larger or using more material or anything. Its literally like,
starting each next layer a bit off from the last, so each layer from
failure point is overhanging the last. Its building each layer from what I
can see as perfectly accurately but in the wrong place by exactly the same
small distance every layer, making a stair step.

I've got an SKR 1.4 turbo after trying a second e3 turbo and then an skr2
(I couldnt get that firmware working right).

I have to get a BL touch then Ill try again, see if its a motor problem or
if its truly the board... but as I said, exact same problem people state.

gwideman, 2021-08-31:
@leech420 

Ah, I misunderstood your original description.

Your comment about 45 degrees was regarding the angle of vertical edges not horizontal edges.  Once the problem occurs, walls that should be vertical start to lean over, due to apparently a progressively increasing layer shift.  

I don't see hot that would be caused by stepper chip registers getting upset per se.

In your original report you said that the printer "stretch the part in Y axis". This could result if the stepper for the Y axis had changed its scaling.  But then in your retelling you say "Because its not doubling steps and making the part
physically larger or using more material or anything."  So maybe it's not stretching the part at all.

You also say that your printer is a Core XY.  If this is so, then having one stepper driver change its scaling would cause stretch in one of the diagonal axes, not the X or Y axis individually.  But then you say it's an Ender 5 plus, which, as I understand it, is not a Core-XY printer.  

So at this point we have a bunch of conflicting information on the problem.

leech420, 2021-08-31:
Im sorry I am not savvy on the core xy argument ive tried to look up
several times what a core xy actually means, no one really makes sense to
me as when they describe it, it sounds just like the ender 5s design.

Sorry if Im not comprehending you correctly I have a migraine right now. I
don't know what is wrong with the printer but Ive tried 2 boards and they
did the same thing. Ive checked all connections, and when I describe what
happens, the symptoms are identical to the ones described here as if I
described them myself. I have no doubt its the same problem, but the
outcome is different. My motors still run fine for a few hours, then they
get lava hot, and suddenly the sounds of the motors get crazy loud like
they are screaming, and the printer continues but building at an angle. I
dont know the terminology for what this problem would be called, except for
that ive got a dozen prints that were great and then suddenly they start
building sideways. Only a power reset makes the motors sound/act normally.

I probably didnt expand on my issue because I didnt want to repeat everyone
else.

Yes you described it exactly right its a vertical shift that suddenly
progresses in one axis (but at a constant rate).

leech420, 2021-08-31:
[image]

please enjoy this image for reference.

gwideman, 2021-08-31:
@leech420 

> My motors still run fine for a few hours, then they get lava hot, and suddenly the sounds of the motors get crazy loud

All of the motors get hot? Including the extruder motor, and the Z axis motor(s)? Or some of them? Or just one? Which ones specifically get especially hot?  Do you have a way to measure their temperature, such as with an IR thermometer?  Are you accustomed to how hot they get when operating normally, say after an hour of normal operation?

In you picture, as you noted, there's a layer shift, indicating that one axis of X and Y is misbehaving. But the Z axis and the other axis of X and Y, and also the extruder, look to be working fine.

The layer shift is conspicuously uniform, like there's something systematic throwing off the frame-of reference in that direction.  However, if one or more motors are getting extremely hot, that's probably the first problem to attack.

leech420, 2021-08-31:
> @leech420
>
>
> All of the motors get hot? Including the extruder motor, and the Z axis motor(s)? Or some of them? Or just one? Which ones specifically get especially hot? Do you have a way to measure [...]

Hi,

Before the fault, the z steppers are cool, only the extruder getting mildly hot. Then all the motors get incredibly hot after the fault, even the z steppers. I dont know how hot but they burn within a second of skin contact so Im thinking 100c. Im not asking you for a solution although I am happy to inform the interested and I hope a more broad fix comes.. I dont know though. Like I said, Im trying the SKR 1.4 turbo, Ill update if I have the same issue just to clean this up on my part. My only other thought is possibly a faulty motor where the windings got overheated as some point in its life, and it causes a short or some sort of discharge as stated once it operates enough to get upto regular temp. Im just spit balling here, will update if I know anything for sure.

leech420, 2021-09-07:
Hello,
So update: New SKR 1.4turb installed and working. Let me say these boards start off silent, and get noisier.. its about 1h30min into a print, and the motors are sounding like ****. All the motors so far feel reasonable, Zs are cold, X is warm, E is warm, and Y is hottest but not burning.

I'm starting to suspect that the drivers arent the issues since my hopes of not imminent failure are not high. within 5h the print will finish and Ill update. If it fails in the same mode Ill update... ugh it sounds like garbage already though... just hasnt gone full blown chainsaw yet.

leech420, 2021-09-07:
> Hello,
> So update: New SKR 1.4turb installed and working. Let me say these boards start off silent, and get noisier.. its about 1h30min into a print, and the motors are sounding like ****. All the [...]

update: Print finished, 6h 30min and looks good, didnt fail. I guess Im sticking with the skr1.4turb as opposed to the e3 turbo board, although I do wonder about the noise, I feel it prints louder that it usually had, but it is also in my room now as opposed to another. None of the motors got particularly hot the entire time.

NRGizzer, 2021-09-12:
I have excatly the same issue. I also reducrd the current of the stepper motors and grounded them but it still happens that at some time they get extremly hot with clacking noices and layer shifting starts.

@leech420 Please report back if the issue did not happend with the SKR 1.4 Turbo, because then i will switch to this board also. Do you already have the problem with the loud steppers with this board?

leech420, 2021-09-12:
> I have excatly the same issue. I also reducrd the current of the stepper motors and grounded them but it still happens that at some time they get extremly hot with clacking noices and layer shifting [...]

Hey.

I replied to your youtube comment. Ive beeb running this board nonstop for a few days now, has been printing flawlessly. Stop using the E3 turbo.

NRGizzer, 2021-09-12:
> Hey.
> 
> I replied to your youtube comment. Ive been running this board nonstop for a few days now, has been printing flawlessly. Stop using the E3 turbo.

Thank you very much for the quick reply.
Do you use the old 1.4 SKR Turbo or the new SKR 2.0 one? And do you plugin TMC 2208 UART Drivers or TMC 2209 (the same as on the e3 turbo)?

leech420, 2021-09-12:
> Thank you very much for the quick reply.
> Do you use the old 1.4 SKR Turbo or the new SKR 2.0 one? And do you plugin TMC 2208 UART Drivers or TMC 2209 (the same as on the e3 turbo)?

good question to ask.
I actually got an SKR2 initiall and gave up trying to make firmware.

Im on the SKR 1.4turbo. Be EXTREMELY certain that you are buying the 1.4 turbo, some amazon sellers and others like to hide multiple search terms in the description, thats actually how I ended up with the SKR2 first. 
Make sure its the 1.4turbo, and Im using tmc 2209 with UART mode. I was able to follow Kersey Fabs tutorials on youtube and it works. Just be extra sure the white and black wires for the BLtouch are correct, because its a little unclear in his video, I burnt one with the skr2.. just be sure to look up a pinout for the SKR 1.4turbo to make sure. I believe black goes to the ground z stop (duh) and the white goes to the signal, not the 5v power in that connection.

NRGizzer, 2021-09-12:
So building the firmware is not a problem. I already built the firmware for my e3 turbo and i just checked - i can succesfully compile the firmware for the SKR 2.0. Would you prefere the SKR 2.0 over the 1.4? And on amazon DE i could get it with TMC 2208 UART or TMC 2209 (without UART). I have to say i dont really understand the difference enough to take the right choice here. Maybe you can help me?

leech420, 2021-09-12:
> So building the firmware is not a problem. I already built the firmware for my e3 turbo and i just checked - i can succesfully compile the firmware for the SKR 2.0. Would you prefere the SKR 2.0 over the 1.4? And on amazon DE i could get it with TMC 2208 UART or TMC 2209 (without UART). I have to say i dont really understand the difference enough to take the right choice here. Maybe you can help me?
> 
> EDIT: Ok, i just ordered the SKR 2.0. Will give it a shot and i hope the heat problem is gone with that version.

I couldnt tell you because I suck at compiling from scratch to fit my printer. Im sure itll be fine if you know what you are doing, It has micro fuses which are a pain to order, so dont burn them out. Also its faster and has some features idk. Another bummer with the Sk1.4turb that i found is common is the neopixel connection doesnt work and has to be wired differently and called out in firmware, I havent gotten to it.

I dont know what UART is lol. I couldnt tell you what to get. I would just get what other people say works, because if it wasnt for that id still be rolling around on my back on the floor trying to get multiple stupid e3 turbos to work.

skiwall, 2021-09-13:
UART is a serial communication protocol. In this instance it used to send data from the microprocessor to the driver chips so you can change the drive current from the LCD instead of using a potentiometer to set a vref. It also swaps the driver mode from stealthchop to spreadcycle. Stealthchop allows for super silent running but spreadcycle will be noisier and is usually activated when larger or faster moves are required. Spreadcycle can be turned of in Marlin, just search for hybrid mode and uncommet. Another advantage of UART is sensorless homing.

NRGizzer, 2021-09-15:
Hey skiwall, much thanks for this explanation :).

By the way - i switched the E3 Turbo against a SKR 2.0 and all the heating problems are gone. I used the exact same marlin firmware settings and my printer is running a large print since 24hours and all temperatures are absolutly stable, no layer shifting or anything like that. With such a hardware issue in my opinion the e3 turbo has to be recalled from selling (dont know how to say that in english, but i think you all know what i mean :-D).

lcortese, 2021-11-03:
Hi guys, how are you?
I am investigating an issue with a SKR E3 v1.2: I buyed that board in 2020, but I did not mounted on a Ender 3, I used it into a Prusa Bear, for this I recicled some parts from an old Ender 3 (Motors, displays, cables, PSU).

It worked fine in the last year, but since some weeks ago the X axis (just this axis) has a issue: in a any momment of the printing, starts to duplicating the logitude. You can see these videos: https://photos.app.goo.gl/AKPnSe8YfbuvLYMn6

Initially I though this was an issue with the gcode or with the SD card, so I replaced it and generated a new gcode, but the problem persist.

I leaved a question in the facebook Bigtreetech group, and a good man said possibly the driver is restarted by an issue (related to ground), and the microsteps pass to be 8 instead of 16.

Instantly I remembered that in the past (sometimes) when I removed the steelshet (after finish the print), the printer was restarted doing a sound like electrostatic.

So I decided to remove and review/repair all the printer cables, also clear the board and the electronic case, then the issue started to happen less times. But still happening.

After a lot of hours of investigation I arrived to this thread.
I did a test connecting the X stepper motor to the doorframe (it should absorb the electrostatic). But it failed again.

So I can say that connecting the stepper to a ground not worked for me :(

if you have any other idea, I can do the test.

leech420, 2021-11-17:
Hey bud.

I hate this board. I switched to the SKR 1.4 turbo and never had a single
issue.

Just do yourself a big fat favor and throw the E3 turbo in the garbage and
dont buy another one. They dont work. If your having issues like what
people are saying, but junk it.

I dont think theres much else to say, these boards are trash and the
countless hours Ive spent on the floor pissing about with the printer with
these boards was hell and time Ill never get back.

Just get another board if youre still struggling.

ur7x, 2023-02-13:
I got one of these boards from old stock... (updated version without the PID power issues)  Installed it in my Ender 5 plus used the firmware that Kersey Fabrications published and I have 4 very hot stepper after about 4 mm of printing.   Added all of the grounds as recommended in this thread, I slowed down the print speed and reduced the current  to the steppers and the problem remains.

Very hot Steppers and layer shifts.    Unless someone has some news for this board, I'm ready to chuck it in the trash. 

One more question, how much heat can these steppers take before they are toast?

[3 more replies]