bigtreetech/BIGTREETECH-SKR-E3-Turbo #34: Huge errors until reboot

yourbuddydinec, 2021-03-01:
So I have been having bad problems to the point things started to melt. My extruders, for some reason, seem to go haywire with the board till I reboot. It happens every other day or so. The first time I noticed was when the extruder mount melted because of the heat it was causing. Nothing would stop this, not even swapping out motors, until I rebooted the machine. Another thing happened a moment ago where the extruder started to make extremely loud noises so i felt it and the gear for the extruder was spinning faster than I ever had seen it before. This again was fixed with a reboot. Could the eeprom be failing? I'm hoping the fix is easy because the amazon seller refuses to help me.

pozi78, 2021-03-01:
I think same problem

https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/31#issue-808118543

yourbuddydinec, 2021-03-02:
Same thing happening to me the last few weeks too. Melted my extruder mount I printed.