bigtreetech/BIGTREETECH-SKR-E3-Turbo #37: BTT SKR E3 Turbo -> creating Ender-5-Pro Firmware with BLTouch not working

foxx-c, 2021-03-18:
Hello,

I'm trying to create an Ender-5-Pro Firmware for my new BTT SKR E3 Turbo mainboard and BLTouch as instructed here:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo

**Did anyone got it working? Is there anyone who can help me? I'm noobie in that topic.**

I'm using VSCode (+ platformIO ide and Auto Build Marlin plugin) and the following setup:

**Firmware 2.0.7.2 from here:**
https://github.com/MarlinFirmware/Marlin
--> either **default** branch or **bugfix**, I tried both with the same result (see below)

I also tried the firmware from the BigTreeTech Github:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/tree/master/firmware/Marlin-2.0.7.2-SKR-E3-Turbo
--> again same result.

**Ender-5-Pro with BLTouch config for BTT SKR E3 Turbo from here:**
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-5%20Pro/BigTreeTech%20SKR%20E3%20Turbo%20with%20BLTouch

or here:

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-5%20Pro/BigTreeTech%20SKR%20E3%20Turbo%20with%20BLTouch

So I tried all the combination but at the end I don't get the firmware compiled.

I always set **default_envs = LPC1769** in platformio.ini file.

I also tried it on a Mac and Windows PC.

Also I tried the Ender-5-Pro config without BLTouch, again same result.

**That's what happens:**

######################################################################

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduino-lpc176x 0.2.6 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/DebugMonitor.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/HAL.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/HAL_SPI.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_flash.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_sdcard.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_wired.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/main.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/tft/tft_spi.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/tft/xpt2046.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/timers.cpp.o
[... 136 lines cut ...]
                 from Marlin/src/HAL/LPC1768/tft/xpt2046.cpp:20:
Marlin/src/HAL/LPC1768/tft/../../../inc/SanityCheck.h:2434:23: error: 'X_SLAVE_ADDRESS' was not declared in this scope
 2434 |   INVALID_TMC_ADDRESS(X);
      |                       ^
Marlin/src/HAL/LPC1768/tft/../../../inc/SanityCheck.h:2432:52: note: in definition of macro 'INVALID_TMC_ADDRESS'
 2432 | #define INVALID_TMC_ADDRESS(ST) static_assert(0 <= ST##_SLAVE_ADDRESS && ST##_SLAVE_ADDRESS <= 3, "TMC2209 slave address must be 0, 1, 2 or 3")
      |                                                    ^~
Marlin/src/HAL/LPC1768/tft/../../../inc/SanityCheck.h:2434:23: error: 'X_SLAVE_ADDRESS' was not declared in this scope
 2434 |   INVALID_TMC_ADDRESS(X);
      |                       ^
Marlin/src/HAL/LPC1768/tft/../../../inc/SanityCheck.h:2432:74: note: in definition of macro 'INVALID_TMC_ADDRESS'
 2432 | #define INVALID_TMC_ADDRESS(ST) static_assert(0 <= ST##_SLAVE_ADDRESS && ST##_SLAVE_ADDRESS <= 3, "TMC2209 slave address must be 0, 1, 2 or 3")
      |                                                                          ^~
*** [.pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_sdcard.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_wired.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/LPC1768/tft/tft_spi.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/LPC1768/main.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/LPC1768/timers.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/LPC1768/tft/xpt2046.cpp.o] Error 1
======================================================================================================================================== [FAILED] Took 3.55 seconds ========================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:03.551
================================================================================================================================== 1 failed, 0 succeeded in 00:00:03.551 ==================================================================================================================================

thrutheframe, 2021-03-22:
Don't use Marlin Auto Build plugin. To build the firmware, look for the thin blue bar on the bottom of  VScode. Click the tick for "PlateformIO:Build"

in configuration.h do ensure that you define motherboard (Line130 - 133)
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_E3_TURBO
#endif

chapito, 2021-12-31:
IM having similar build issues with Ender3 Pro with upgraded BTT SKR E3 Turbo MB. (will try Thrutheframe's hit)