bigtreetech/BIGTREETECH-SKR-E3-Turbo #38: issue with Mixing extruder

juavinas, 2021-03-20:
Hello,

I am trying to do it mixing extruders in Zsyong 2in1 head. 1Head & 2 extruders. Printer Ender 3  #pro.
in my Configuration.h:

#define EXTRUDERS 2
#define SINGLENOZZLE

#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
  #define GRADIENT_MIX           // Support for gradient mixing with M166 and LCD
  #if ENABLED(GRADIENT_MIX)
    //#define GRADIENT_VTOOL       // Add M166 T to use a V-tool index as a Gradient alias
  #endif
#endif

& error from Vscode are: 
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1001:6: error: #error "For MIXING_EXTRUDER set MIXING_STEPPERS > 1 instead of EXTRUDERS > 1."
 1001 |     #error "For MIXING_EXTRUDER set MIXING_STEPPERS > 1 instead of EXTRUDERS > 1."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1001:6: error: #error "For MIXING_EXTRUDER set MIXING_STEPPERS > 1 instead of EXTRUDERS > 1."
 1001 |     #error "For MIXING_EXTRUDER set MIXING_STEPPERS > 1 instead of EXTRUDERS > 1."

Attaching Configuration files.
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/6176436/Marlin.zip)

radek8, 2021-06-21:
Set
#define EXTRUDERS 1

hiskynet, 2022-04-04:
high have same prob

if define EXTRUDERS to 1 then lose options to control define EXTRUDERS move