bigtreetech/BIGTREETECH-SKR-E3-Turbo #39: True Dual Z using E1 Stepper driver

TheGunPsy, 2021-04-10:
Hi, 
What is the best Marlin 2.0 firmware configuration for utilizing the E1 stepper driver to run a second Z stepper motor? Due to how new this board is, I am not finding many resources on this. I would really like to learn how to do this. 

I am running the Ender 3 w/ BL Touch and just bought this board. I am trying to use the Z auto align feature and it appears I need 2 Z stepper drivers in order for it to work. I know basic coding and am able to configure marlin pretty well. My confusion lies in that if using the E1 stepper, wont marlin treat the motor like it is extruding vs lifting on Z? 

Thanks for any input in advance.

thisiskeithb, 2021-04-10:
I setup some config options for an Ender-3 and this board that should get you going: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo

TheGunPsy, 2021-04-10:
> I setup some config options for an Ender-3 and this board that should get you going: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo

Thisiskeithb, 
Thanks for the info. I inspected your config.adv file and on line 648 I noticed that you have 2 Z steppers uncommented. Now, on line 821, I noticed that Z stepper auto align is turned off. This is the feature I am trying to use. Since line 648 is turned on, my hardware configuration would use E1 to power the Z2 stepper, and to turn on auto align, I would just need to uncomment 821? 

Thanks!

thisiskeithb, 2021-04-10:
> I would just need to uncomment 821?

Yep.

SpectrumTech, 2021-04-28:
I have been trying to get this to work for months now... If I enable z-stepper auto align, E0 stops working and the printer will not extrude or other problems appear. I’m curious to find out if this is actually working or not. I have pretty much given up on this board working right on a dual-z printer.

thisiskeithb, 2021-04-28:
@SpectrumTech: Here’s a link to my personal E3 Turbo config with dual Z that works: https://github.com/thisiskeithb/Marlin/tree/myconfig/E3-SKR-E3-Turbo-Dual-Z-BLTouch

The readme has full details of the hardware addons.

LearnIIBurn, 2021-04-28:
Dude @thisiskeithb what a stand up guy you are! :) Thanks for sharing, it's really cool to have a Marlin developer so hands on with the community like you are.

thisiskeithb, 2021-04-28:
This PR fixes a non-boot issue with an SKR 2: https://github.com/MarlinFirmware/Marlin/issues/21732

thisiskeithb, 2021-04-28:
Glad to help!

TheGunPsy, 2021-06-07:
> I setup some config options for an Ender-3 and this board that should get you going: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo

Hey Thank you very much. I have been busy with other things. I am in the process of cutting the new firmware and just got dual z installed. I will let you know how i fare. Thanks again!

closed 2021-06-12 by TheGunPsy