bigtreetech/BIGTREETECH-SKR-E3-Turbo #40: DWIN display Ender 3v2

tome9111991, 2021-05-12:
Hi
Has anyone an idea how to get work the E3 turbo with Ender 3v2 dwin display?

tome9111991, 2021-05-15:
E3 turbo dont have tx & rx pin?

radek8, 2021-06-12:
[image]

tome9111991, 2021-06-13:
Ok.
But i have still issue to compile firmware.
[image]

radek8, 2021-06-13:
Send configuration files

radek8, 2021-06-13:
It will probably be better to disconnect the original Ender 3 V2 display and replace it with a BTT TFT display.
There is no documentation available for the original Ender display and it is difficult to determine which pins are to the serial port.

badbod, 2021-06-18:
for the original V2 display uncomment : 
#define DWIN_CREALITY_LCD

and comment out any other displays

There is no serial connector on this display, just use the original cable and plug into EXP1.

There will be a bunch of features you can't use though. As suggested above, maybe get a Creality CR-10 12864 (cheap), and  uncomment  : 
#define CR10_STOCKDISPLAY
#if ENABLED(CR10_STOCKDISPLAY)
  #define RET6_12864_LCD  // Specific to the SoC (can either be RET / VET)
#endif
and you get all the Marling features enabled.
No need for serial either just use the original cable in EXP1 slot on the 12864 board,

  OR maybe even something like the BTT TFT35 E3 V3  which has both he CR-10 mode and touch display mode. Uses the EXP1 AND  serial connections for double the fun  :-)  Plus some other stuff like USB thumb drive support, extra serial ports and SUPER LOUD BUZZER/SPEAKER,  really, I mean it,, dont take the sticker off the speaker unless you want to annoy the neighbours and make your super expensive HIFI look wimpy and lacking in watts.

tome9111991, 2021-06-18:
It works.

[image]

badbod, 2021-06-18:
Hurrah

badbod, 2021-06-18:
also wondering why you chopped up the original cable,?    it is long and needs all the connections for LCD usage

badbod, 2021-06-18:
does the original cable not work if you just plug it in?   mine worked fine,  I did read some boards had the connector wrong way round so you had to either cut a slot on the other side of the socket on the board so you could plug the cabkle in the other way round.  I did not have to do that though,   hmm,  just wondring why you got the loose wires when it should work fine with the original cable 'unmolested'  :-)

tome9111991, 2021-06-18:
I have torewire the Original displaycable.
I put jumper cable in the 10pin port 

I use jumper cable just for test.
I need to make "new" displaycable

badbod, 2021-06-18:
The original  V2 cable will work fine, no need to rewire.

badbod, 2021-06-18:
if the original V2 cable does not work, then cut off the plastic part that stops you plugging it in the wrong way round,  then plug it in the wrong way round.  It will work then.

itsround, 2021-06-19:
@tome9111991 what have you changed in your code? I am still running into errors while compiling. I am using Ender 3 V2. Uncommented #define DWIN_CREALITY_LCD and commented #define CR10_STOCKDISPLAY in my configuration.h. Compiling for SKR E3 Turbo still fails.

tome9111991, 2021-06-19:
> @tome9111991 what have you changed in your code? I am still running into errors while compiling. I am using Ender 3 V2. Uncommented _#define DWIN_CREALITY_LCD_ and commented _#define CR10_STOCKDISPLAY_, _#define LCD_BED_LEVELING_ in my configuration.h. Compiling for SKR E3 Turbo still fails.

That is right.

You use 
`#define SERIAL_PORT 3 
//#define SERIAL_PORT_2 -1`

and add in
Pins/lpc1769/e3 turbo
`Change some pin like:
#if ENABLED(DWIN_CREALITY_LCD)

#ifndef BEEPER_PIN
#define BEEPER_PIN               EXP1_10_PIN
#undef SPEAKER
#endif

#define BTN_EN1                  EXP1_03_PIN
#define BTN_EN2                  EXP1_04_PIN
#define BTN_ENC                  EXP1_06_PIN

#endif`

In
Inc/Conditionals_LCD.h
Search: DWIN_CREALITY_LCD

Change  `#define LCD_SERIAL_PORT 3` to
`#define LCD_SERIAL_PORT 1`

itsround, 2021-06-20:
@tome9111991: thank you! Building the .bin worked!

One question regarding the serial ports.
If I change `#define SERIAL_PORT` from `-1` to `3`
and comment `#define SERIAL_PORT_2`

can I still use the USB port for my raspberry with Octoprint?

tome9111991, 2021-06-20:
I dont know
I have no raspberry

itsround, 2021-06-22:
@tome9111991 After flashing the .bin-file my display makes a a loud continuous tone. Must I do some adjustment on the pinning?

tome9111991, 2021-06-22:
You have to rewire your displaycable

**Display -> Board**
Orange -> Blue
Blue -> Brown
Yellow -> Red

itsround, 2021-06-22:
@tome9111991 thanks for your help!
This are the pins. 1 and 2 are clear.
3 (Ender) --> 8 (SKR)
4 --> 6
5 --> 10
6 --> 9
am i right so far?
Whats with TX and RX?

           Ender 3 V2 display                              SKR E3 TURBO 
                 _____                                           _____                          
             5V | 1 2 | GND                                  5V | 1 2 | GND               
    (BTN_E1) A  | 3 4 | B (BTN_E2)               (LCD_EN) P0_18 | 3 4 | P0_17 (LCD_RS)          
           BEEP | 5 6   ENT (BTN_ENC)            (LCD_D4) P0_15 | 5 6   P0_20 (BTN_EN2)              
             TX | 7 8 | RX                                Reset | 7 8 | P0_19 (BTN_EN1)               
             NC | 9 10| NC                      (BTN_ENC) P0_16 | 9 10| P2_08 (BEEPER)                  
                 -----                                           -----

tome9111991, 2021-06-22:
Ender --> SKR
8 --> 5
5 --> 10
7 --> 9

On E3 turbo P0_15, P0_16 are tx,rx with `LCD_Serial_Port 1`

itsround, 2021-06-24:
> Ender --> SKR
> 8 --> 5
> 5 --> 10
> 7 --> 9
> 
> On E3 turbo P0_15, P0_16 are tx,rx with `LCD_Serial_Port 1`

Thank you! It works 👍🏼

badbod, 2021-06-27:
Strange, I followed the documentation, which states if your using the original V2 display, plug it in using original cable. I made no changes to pins and cables wiring etc. it works just fine.  Using the configuration from 
   Configurations-release-2.0.9\config\examples\Creality\Ender-3\BigTreeTech SKR E3 Turbo\

uncomment 
#define DWIN_CREALITY_LCD

comment
//#define CR10_STOCKDISPLAY

compile build and flash.
BTT documentation said it would, and it did, so I dont understand all the rewiring and playing with pins file. 

not all BTT SKR-E3-Turbo the same I guess?

tome9111991, 2021-06-28:
@badbod 
can't work with Configurations-release-2.0.9\config\examples\Creality\Ender-3\BigTreeTech SKR E3 Turbo\

becauce LCD_BED_LEVELING dont work with e DWIN_CREALITY_LCD
and dwin is looking for defined pin: BEEPER_PIN, BTN_ENC, ....
if not define you got a error

jonasl03, 2021-09-19:
I've tried basically everything, and can't get this to work for the love of my life. All I get is a black screen.
Running the newest marlin build, and I've rewired it according to pins_BTT_SKR_E3_TURBO
Any clue what the hell is up?

tome9111991, 2021-09-19:
In latest marlin bugfix build you dont need change.

Only in pins_BTT_SKR_E3_TURBO

`#error "Ender-3 V2 display requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue."`

to

`//#error "Ender-3 V2 display requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue."`

jonasl03, 2021-09-19:
I've tried that as well, aka using the default cables. All the screen does is turn on and give a black screen while beeping like crazy. I've also tried re-flashing the screen and printer firmware several times

jonasl03, 2021-09-19:
Maybe my screen is just fucked?

tome9111991, 2021-09-19:
No you have to make a special cable.
Like
[image]

Dont touch black and white wire!

jonasl03, 2021-09-19:
As mentioned before, I have tried rewiring it according to that. All I get is a black screen and beeping unfortunately

tome9111991, 2021-09-19:
You got black screen/beeping because your pins/wire is wrong.

You can post your config?

jonasl03, 2021-09-19:
Gonna send a picture of my wiring in a second

`#if ENABLED(DWIN_CREALITY_LCD)
  //#error "DWIN_CREALITY_LCD requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue."

 /**
  *          Ender 3 V2 display                       SKR E3 Turbo (EXP1)                Ender 3 V2 display --> SKR E3 Turbo
  *                ______                                     ______                                  RX  8 -->  5  P0_15
  *            5V | 1  2 | GND                            5V | 1  2 | GND                             TX  7 -->  9  P0_16
  *   (BTN_E1)  A | 3  4 | B   (BTN_E2)       (LCD_EN) P0_18 | 3  4 | P0_17 (LCD_RS)              BEEPER  5 --> 10  P2_08
  *        BEEPER | 5  6   ENT (BTN_ENC)      (LCD_D4) P0_15 | 5  6   P0_20 (BTN_EN2)
  *  (SKR_RX1) TX | 7  8 | RX  (SKR_TX1)               Reset | 7  8 | P0_19 (BTN_EN1)
  *            NC | 9 10 | NC                (BTN_ENC) P0_16 | 9 10 | P2_08 (BEEPER)
  *                ------                                     ------
  */

  #define BEEPER_PIN                 EXP1_10_PIN
  #define BTN_EN1                    EXP1_03_PIN
  #define BTN_EN2                    EXP1_04_PIN
  #define BTN_ENC                    EXP1_06_PIN
`

jonasl03, 2021-09-19:
Gonna send a picture of my wiring in a second

Do you want the pins_BTT_SKR_E3_TURBO.h?

tome9111991, 2021-09-19:
In latest marlin verson you dont need a change is already in.

Only need
Special cable
`#define SERIAL_PORT  3`
`//#error "Ender-3 V2 display requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue."` in pins_BTT_SKR_E3_TURBO.h

thats all an works well for me.

jonasl03, 2021-09-19:
My config is default, I haven't touched it. I know it has what it needs in the latest marlin version
I've rewired my cable according to what is stated in this thread

`Ender 3 V2 display --> SKR E3 Turbo
                 8 --> 5
                 5 --> 10
                 7 --> 9`

jonasl03, 2021-09-19:
[image]
[image]
[image]

tome9111991, 2021-09-19:
looks a little confusing😅
You are sure is correct?

I guess your green wire from display should plug in your current yellow pin for exemple.

Here my jumper cable
[image]
[image]

jonasl03, 2021-09-19:
I used that last image, which was the issue
Numbers on it aren't in the right order
But I got it working now thanks to the images of your jumper cables, thanks bud.

SISLANGER, 2021-10-08:
@tome9111991 thank you will give it a go.

erdoganguner, 2021-11-26:
hey so can anyone send the config files with the "fixed" firmware?

tome9111991, 2021-11-27:
Config:
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2/BigTreeTech%20SKR%20E3%20Turbo/CrealityUI

Firmware:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

erdoganguner, 2021-11-27:
It gives me this error when I try to compile it

`Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
---------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\users\guner\documents\marlin-2.0.x\marlin\src\pins\pins.h:413,
                 from c:\users\guner\documents\marlin-2.0.x\marlin\src\inc\marlinconfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\guner\documents\marlin-2.0.x\marlin\src\pins\lpc1769/pins_BTT_SKR_E3_TURBO.h:205:4: error: #error "Ender-3 V2 display requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue."
  205 |   #error "Ender-3 V2 display requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue."
      |    ^~~~~
CalledProcessError: Command '"C:\Users\guner\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -D__MARLIN_FIRMWARE__ -DNDEBUG -DU8G_HAL_LINKS -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\guner\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\guner\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\guner\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\guner\Documents\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 320:
    apply_features_config()
  File "C:\Users\guner\Documents\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 139:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\guner\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658: 
    return self.method(*nargs, **kwargs)
  File "C:\Users\guner\Documents\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 296:
    load_marlin_features()
  File "C:\Users\guner\Documents\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 284:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\guner\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\guner\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)`

Do I have to make other changes in platformio? So far I only changed the enviroment to LPC1769

erdoganguner, 2021-11-27:
Since I'm in a bit of a hurry, would it be too much to ask just for the working bin file compatible with the DWIN display?

tome9111991, 2021-11-27:
You have to comment out in pins_BTT_SKR_E3_TURBO.h
Like:

```//#error "Ender-3 V2 display requires a custom cable with TX = P0_15, RX = P0_16. Comment out this line to continue." ```

erdoganguner, 2021-11-27:
I think I can handle modifying the display cable
really thankful by the way

erdoganguner, 2021-11-28:
I don't think I've rewired the cable correctly. Has anyone gotten it working with dupont connectors?
Tried following the photos above; didn't work

tome9111991, 2021-11-28:
Still works for me.
> 
> Here my jumper cable [image] [image]
>

I-dont-want-peace, 2021-12-12:
For some reason I found the above images and the text art confusing. I had to be patient to sort this out but here is, what I think, is a helpful couple pictures from when I did this. I disassembled the screen side connector and clipped the wires that were not being rearranged about 3mm, the ones that I was moving about 1mm (just enough to refresh the insulation) and then used to super thin screw driver to start them into the connection blades and then used the first piece of the connecter to press them back into their connections during reassembly. The orientation nub is hidden by the wires and under my index finger.

Works great now.
[image]
[image]

PS.  If you're also connecting a BLTouch ensure you have the wiring correct. I had them wrong and the screen would not work (flashing & blank) and the Octoprint Rpi disconnected from Wifi as soon as it booted. Fixing the BLTouch enabled them both to work.

tome9111991, 2021-12-12:
[image]
 
is based on:
```

 /**
  *          Ender 3 V2 display                       SKR E3 Turbo (EXP1)                Ender 3 V2 display --> SKR E3 Turbo
  *                ------                                     ------                                  RX  8 -->  5  P0_15
  *            5V | 1  2 | GND                            5V | 1  2 | GND                             TX  7 -->  9  P0_16
  *   (BTN_E1)  A | 3  4 | B   (BTN_E2)       (LCD_EN) P0_18 | 3  4 | P0_17 (LCD_RS)              BEEPER  5 --> 10  P2_08
  *        BEEPER | 5  6   ENT (BTN_ENC)      (LCD_D4) P0_15 | 5  6   P0_20 (BTN_EN2)
  *  (SKR_RX1) TX | 7  8 | RX  (SKR_TX1)               Reset | 7  8 | P0_19 (BTN_EN1)
  *            NC | 9 10 | NC                (BTN_ENC) P0_16 | 9 10 | P2_08 (BEEPER)
  *                ------                                     ------
  */
```

Nuttavoot, 2022-01-12:
Could you show me the photos of the screen when using this board with original v2 LCD ?

Which ui you’re using?

tome9111991, 2022-01-12:
> Could you show me the photos of the screen when using this board with original v2 LCD ?
> 
> Which ui you’re using?

what do you want to see? there is no difference to the stock board.

I'm use EnhancedUI cirrently but the other UI's ist working too.