bigtreetech/BIGTREETECH-SKR-E3-Turbo #41: EEPROM Write Fails

Z3r0Auth, 2021-06-14:
Built Marlin 2.0.7.2 from this distro for this board, and have not been able to write to EEPROM successfully, receiving the following output firing M500:  
  
Send: M500
Recv: echo:Error writing to EEPROM! [Repeats]
Recv: echo:Error writing to EEPROM!
Recv: echo:Settings Stored (3 bytes; crc 7920)
Recv: Error:EEPROM datasize error.
Recv: echo:Error writing to EEPROM!
Recv: ?Unable to save mesh data.
Recv: ok

No changes made to pins file or config in regards to EEPROM config.

Z3r0Auth, 2021-06-14:
Just got this board, and have been building it into my Ender5+ and while i've got everything seeming to work fine, I can't save any settings.  
  
Marlin 2.0.7.2 built using the files from the linked distro specifically from BTT, modified only the needed settings to drive the steppers the right way, trigger/stow/read the BLTouch.. but went to write a few settings into storage with M500 last night after running first successful UBL and it just pours EEPROM write errors.  
  
This one's getting past my depth on this.  
  
configuration.h: https://pastebin.com/fuJKJQDx  
configuration_adv.h: https://pastebin.com/PBK4DMDL
pins_BTT_SKR_E3_TURBO.h: https://pastebin.com/S6yPS2Gv

scottjvincent, 2021-06-15:
I had the exact same problem with my ender 3 v2 , after I upgraded and went with the newest Marlin build it saves to epromm no problem

Z3r0Auth, 2021-06-15:
> I had the exact same problem with my ender 3 v2 , after I upgraded and went with the newest Marlin build it saves to epromm no problem

did you go to a nightly/bugfix build, or 2.0.8.2 (current main release)?

scottjvincent, 2021-06-15:
I did the 2.0.8.2 main build..... Tried absolutely everything first.... Ever firmware setting I could think/read of...... And finally installed the current version and went though the entire firmware beginning to end and changed all the needed settings in  vs code with the platformI0 plug-in installed in it..... Which was completely new to me... And make sure all the epromm settings are enabled in the firmware and then all the settings would save to epromm correctly.....
Definitely more work then I wanted it to be for sure, but I finally got it working....

Scott V.

Z3r0Auth, 2021-06-15:
I took @scottjvincent 's advice and completely built a new config on 2.0.8.2 and was successful in getting it to reset, write, and recall EEPROM data, so that's resolved my current issue personally.  
  
It was suggested on Reddit that there's a known issue with this distro, referencing this ticket fo anyone who want's to try and fix this one:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/16  
  
Seems like this fix is already in the mainline Marlin though, so i'd say it's worth the effort to port forward into new configs.  
  
I just apparently have great luck, in that Marlin 2.0.9 rolled out the day after I built an entirely new config set :|  
  
Thanks @scottjvincent for at least confirming the effort would be worth it

radek8, 2021-06-16:
The new version of Marlin is fixed.
The fix for the older version is here
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/issues/30

scottjvincent, 2021-06-16:
Ya I tried that also with the older version and it didn't work for me.

radek8, 2021-06-16:
@Z3r0Auth 
All you have to do once is create the configuration and then it is never a problem to upgrade to the new version if you are using the correct SW :-)
https://desktop.github.com/

Simply download the new PR and your settings will take effect

badbod, 2021-06-18:
I notice that Creality disabled eeprom in thier 2.0.7.2 build.  First time I used eeprom was when I compiled my own 2.0.7.4 or whatever the release was from Marling github. Was suprised I no longer needed the SDcard to save settings  :-)    just make sure #define EEPROM is uncommented and it should work fine. 

Have never even tried compiling the 2.0.7.x releases.  but I guess it does not work, or Creality really wanted you to carry your settings round on the sdcard. I lost all my settings a few times unplugging the card without switching the printer off.  No matter, those days are gone.
 
Using the Marlin 2.0.9 now and all functions working fine with my Ender 3 V2 BLtouch, dualZ, auto Z leveling, UBL 10x10 mesh linear advance direct drive dual extruders  etc etc.  All the wizards seems to work great and save tons of time leveling the bed and setting Z offset etc. 

Play with the pins and you can setup the onboard thermister as chamber to display along with the nozzle and bed  :-)

scottjvincent, 2021-06-18:
> @Z3r0Auth
> All you have to do once is create the configuration and then it is never a problem to upgrade to the new version if you are using the correct SW :-)
> https://desktop.github.com/
> 
> Simply download the new PR and your settings will take effect
Really??

> @Z3r0Auth
> All you have to do once is create the configuration and then it is never a problem to upgrade to the new version if you are using the correct SW :-)
> https://desktop.github.com/
> 
> Simply download the new PR and your settings will take effect

That desktop GitHub program makes upgrading firmware that easy huh?
Cuz making a new config program every time is a pain.

radek8, 2021-06-18:
Yes, updating Merlin is easy then.
It took me a while to understand how it works.
But now it's easy.
You clone the repository (Merin), create your own branch, click your configuration into the branch.
Then all you have to do is merge the new changes, check what's new, or remove the conflicting changes and have the new Merlin ready for compilation.