bigtreetech/BIGTREETECH-SKR-E3-Turbo #44: sensorless homing not working

tome9111991, 2021-06-19:
Hi there.

X and Y axis sensorless homing does not work.

The bed does not stop moving along the x and y axis when homing. I tried cranking the sensitivity all the way up to 255 but it still wouldn't stop at all. Any advice? 

Jumper for X and Y Diag are in

badbod, 2021-06-20:
you have enabled steathchop and sensorless homing in the firmware?

tome9111991, 2021-06-20:
Yes maybe i found the issue.
I comment 
```
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
```
it works now but not well.

With 90 STALL SENSITIVITY he moves a little bit hard to endstop.
With 91 STALL SENSITIVITY he moves 2-3mm and then is homed but not reach the endstop.

Im back to Sensor homing.

closed 2021-06-27 by tome9111991