bigtreetech/BIGTREETECH-SKR-E3-Turbo #45: Heaters turn off and motors stay on and get HOT

scottjvincent, 2021-06-24:
So I recently installed the E3 turbo on my Ender 3 V2 and it has been working great everything works until I tried to print this carbon fiber filled polycarbonate blend.... 90 c bed temperature and 260c nozzle temperature..... Printed two or three things and everything was fine but then on the last one I came out and looked at the printer and the motors were humming it had stopped printing the heaters were off but the motors were just hotter than a bullet..... Don't know why the heaters turned off out of the blue ,using octoprint it says thermal runaway and you can see on the graph the bed heater and the nozzle heater just start cooling down out of the blue..... BTT TFT50 screen says thermal runaway also needs printer reset... But the motors are still active and super super hot.....
I don't have a clue what's going on...
I read the other post that talks about ESD.... But so far it seems everything is fine until I try to print with higher bed and nozzle temperatures than normal..... It's happened two times now and both times were printing the polycarbonate....

First thoughts is is there a way to go in the firmware and make it so the motors turn off if it faults out for  thermal run away???
That doesn't fix the problem but at least it won't burn my printer up when it does fault out.

I have the Ender 3 V2 silent fan upgrade so the PSU and the motherboard have 92 mm fans, so I'm assuming the motherboard isn't overheating??
And no I have not checked grounds on any of the motors yet because it seems like it's only doing it when printing that polycarbonate.....

Any help would be great
Thanks
Scott V.

scottjvincent, 2021-06-24:
So I just made the firmware changes to show chamber temperature which would be the motherboard temperature turned the stepper motor amps down from the default 800mAh to 600 mAh
And change the bed thermal runaway settings up to 40 seconds and 4° swing...... Don't think that last one's going to do anything but since I was there I thought I'd try it...

So now I can at least see how hot the motherboard gets......
What temperature would be overheating for the motherboard?

Thanks
Scott V.

scottjvincent, 2021-06-26:
So i got it to do it again and the terminal in octoprint says this.

Recv: E driver error detected: 0x1075773584.0000000000000000
Recv: coil short circuit
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		true	true	true	true
Recv: Set current	800	800	800	800
Recv: RMS current	1436	1436	1436	1436
Recv: MAX current	2025	2025	2025	2025
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual	31/31	31/31	31/31	31/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	true	true	true	true
Recv: msteps		8	8	8	8
Recv: interp		true	true	true	true
Recv: tstep		667	667	21344	7391
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum	136	79	125	36
Recv: pwm scale auto	78	24	89	0
Recv: pwm offset auto	36	36	36	36
Recv: pwm grad auto	61	62	15	14
Recv: off time	3	3	3	3
Recv: blank time	36	36	36	36
Recv: hysteresis
Recv:  -end		-3	-3	-3	-3
Recv:  -start		6	6	6	6
Recv: Stallguard thrs	0	0	0	0
Recv: uStep count	415	590	96	57
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	70	86	2	6
Recv: stst		*	*	*	*
Recv: olb			*		*
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa					*
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x40:1F:00:00
Recv: 		Y	0x40:1F:00:80
Recv: 		Z	0x40:1F:00:00
Recv: 		E	0x40:1F:00:90
Recv: 
Recv: 
Recv: echo:Driver error
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error"
Send: M112
Send: N2983 M112*17
Send: N2984 M104 T0 S0*22
Send: N2985 M140 S0*83
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

Any ideas?

LTSpenny, 2021-07-14:
Hey so I’m running an ender 3 max and just had an issue where my motors got hot as well. They then stopped being able to torque or turn if there’s any resistance. Happened out of the blue after printing for a few weeks. Brief for my z was at 560 running duals. If I find out anything else I’ll follow up.

radek8, 2021-08-10:
Can you turn off the printer jerk, and enable Junction deviation?