bigtreetech/BIGTREETECH-SKR-E3-Turbo #49: Extruder fan (FAN1_PIN) AUTO_FAN functionality not working

IceDragonix, 2021-08-01:
Hello,

According to a tutorial - https://youtu.be/LxQWCBE6lLk?t=441 - the SKR-E3-Turbo should have FAN1 header automatic on/off feature based on hotend temperature.

I have defined the extruder fan as FAN1_PIN and set EXTRUDER_AUTO_FAN_TEMPERATURE to 50 degrees. 

Unfortunately the fan is always on and never turns off. I compiled Marlin-bugfix-2.0.x from yesterday (07/31/2021). I also tried installing the stock firmware. Also no success.

Have anyone experienced similar behavior? Is there any solution?

```
/**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN FAN1_PIN
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define E6_AUTO_FAN_PIN -1
#define E7_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1
#define COOLER_AUTO_FAN_PIN -1
#define COOLER_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
#define CHAMBER_AUTO_FAN_TEMPERATURE 30
#define CHAMBER_AUTO_FAN_SPEED 255
#define COOLER_AUTO_FAN_TEMPERATURE 30
#define COOLER_AUTO_FAN_SPEED 255

#define TEMP_SENSOR_CHAMBER 0

#if TEMP_SENSOR_CHAMBER
  //#define HEATER_CHAMBER_PIN      P2_04   // Required heater on/off pin (example: SKR 1.4 Turbo HE1 plug)
  //#define HEATER_CHAMBER_INVERTING false
  //#define FAN1_PIN                   -1   // Remove the fan signal on pin P2_04 (example: SKR 1.4 Turbo HE1 plug)

```

IceDragonix, 2021-08-25:
Hey @radek8,

I appreciate the support. Unfortunately, it did not work.

The FAN1_PIN was already set to P2_02 in file Marlin\src\pins\lpc1769\pins_BTT_SKR_E3_TURBO.h

JPU571, 2021-09-11:
I'm having the same issue with this board on current marlin.

radek8, 2021-09-11:
Looking at the wiring diagram, I wonder if there may be a fault in the wiring of the board.
FAN1 connection is different from other MOSFET ...

[image]

JPU571, 2021-09-11:
YouTubers are setting this up and it's working for them. Maybe bug in this marlin version?

radek8, 2021-09-11:
It also works for me, but I have another board (SKT 1.4 Turbo) and I use an external MOSFET to switch.
There will be no mistake in Marlin.

JPU571, 2021-09-11:
So basically, it works for you on another board, with an external component.

radek8, 2021-09-11:
Yes, my board does not only have one adjustable (switched) fan output connector.
I have two hotends, 2 extruder cooling fans that switch above 50 ° C, a switched electronics fan that switches when the stepper motors are running, and one fan to cool the printed object.
I switch 3 fans with external MOSFET transistors. Everything works as it should.

radek8, 2021-09-11:
If you have a well-configured marlin, and yet your fan runs continuously, the PL4009 transistor on the FAN2 connector may be burned out.
Can you measure the level of the signal that switches the transistor?

JPU571, 2021-09-11:
My fan isnt turning on at all. Even with temp change.

radek8, 2021-09-11:
If the transistor burns out, it can be permanently closed or open.

Do you have set up and uncommented in Marla?

#define E0_AUTO_FAN_PIN FAN1_PIN
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed

If so, measure the voltage on the pin I marked. at a temperature below and above the set value

radek8, 2021-09-11:
Attention
In the file
pins_BTT_SKR_E3_TURBO.h
comment the line
// # define CONTROLLER_FAN_PIN FAN1_PIN

radek8, 2021-09-11:
> Hey @radek8,
> 
> I appreciate the support. Unfortunately, it did not work.
> 
> The FAN1_PIN was already set to P2_02 in file Marlin\src\pins\lpc1769\pins_BTT_SKR_E3_TURBO.h

@IceDragonix, Pin1 is already defined in the pin set for the fan-controller, not for cooling the extruder
Try commenting on it

JPU571, 2021-09-11:
That did not work either.  Guess I'm going back to fan being always on for a while.

JPU571, 2021-09-11:
This guy says to use FAN1_PIN for the hotend fan and in his video it works.  
https://www.youtube.com/watch?v=4xtsJ-3-c_U

JPU571, 2021-09-11:
This guy reconfigured the E1 pin to be the auto-fan for E0 to perform this task.  I'm about to try this now.

https://www.youtube.com/watch?v=KFLSpp5ClUY&t=731s

JPU571, 2021-09-11:
This works.. but only after I found that I had a bad connection to my fan.. bad connection was caused by me extending the cables [ Ender Extender ] and the "self-crimping" connectors were not biting into the cables correctly. I'm trying the FAN1_PIN again to see if the cables was the issue there.

JPU571, 2021-09-11:
moved everything back to FAN1_PIN and it works as expected for me, after cabling issue was resolved. Fan activates at 50C, turns off at 49C.

IceDragonix, 2021-09-12:
On my side I never got the FAN1_PIN (P2_02) to work. It's simply always on. Perhaps my board is faulty.

I used the workaround from the video posted by @JPU571 (https://www.youtube.com/watch?v=KFLSpp5ClUY&t=731s) and it works perfectly fine. Instead of using FAN1_PIN (P2_02), we use pin P2_04, which by default is EXTRUDER 1. Obviously, the fan needs to be physically connected to the EXTRUDER 1 on the board.

There is a follow up to the video posted by @JPU571, which explains how to use P2_02 as a motherboard cooling fan, with P1_30 configured as TEMP_CHAMBER_PIN - https://www.youtube.com/watch?v=R6ZGVY3HvQg. Again, this does not work for me.

IceDragonix, 2021-09-12:
> > Hey @radek8,
> > I appreciate the support. Unfortunately, it did not work.
> > The FAN1_PIN was already set to P2_02 in file Marlin\src\pins\lpc1769\pins_BTT_SKR_E3_TURBO.h
> 
> @IceDragonix, Pin1 is already defined in the pin set for the fan-controller, not for cooling the extruder
> Try commenting on it

@radek8 Thank you Radek. I tried that, and no success. P2_02 is always on, no matter what I try.

radek8, 2021-09-12:
It looks like a faulty transistor on pin P2_02

tome9111991, 2022-01-11:
Hi
I have tried on my E3 Turbo
I just add `#define E0_AUTO_FAN_PIN FAN1_PIN` it works.
Only tested with latest Bugfix from today.

Maybe you can re-check your hotend fan connection to Fan1