bigtreetech/BIGTREETECH-SKR-E3-Turbo #63: Stepper Error, error, error. All the time!

DragonFlyer007, 2021-11-20:
Why, always geting this... either it is on Z or E or Y, seems random. But same error... so sick of it. So much wasted print time and plastic!! 

Two brand new ender 5 plus with klipper, and two new E3 turbo cards.

Just about had enough of this now.

[image]

ellensp, 2021-11-26:
GSTAT – Global status flags
1: Indicates that the IC has been reset since the last read access to GSTAT

Ie you had a power glitch

BreeOge, 2021-11-26:
I have marlin and its doing the exact same thing, I had all 3 axis do it at the same time. I have my power in being monitored by a external source.  It shows no power issues, so it has to be board related, or power related beyond incoming utility power.  From other forums I am reading it seems to happen on the SKR Mini as well, I wonder if we have a board issue that has not been found.

BreeOge, 2021-11-28:
The fact that NONE of these issues have a official response or close, makes me think they really really and I mean REALLY dont care about the customer.

DragonFlyer007, 2021-11-29:
Ok, so i have been doing some testing for the last month or so, and the only pointer i'm getting so far i that when i disssable or just mount it elswhere and always leave some filament in the sensor/switch, the errors seams to stop.

The moment i try to use the sensor again, errors are starting to appear within a day or so.

dzervas, 2021-12-02:
The drivers have a "standby" pin that has to be pulled low early in order to function. Check it out on your firmware (klipper/marlin) for the required configuration.

DragonFlyer007, 2021-12-02:
> The drivers have a "standby" pin that has to be pulled low early in order to function. Check it out on your firmware (klipper/marlin) for the required configuration.

Hmm. A google search regarding this have none good results. Explain please? Running Klipper here.

dzervas, 2021-12-02:
Check this: https://github.com/Klipper3d/klipper/issues/3549#issuecomment-739323618

The ticket more or less explains what's happening and how you can solve it

DragonFlyer007, 2021-12-02:
> Check this: [Klipper3d/klipper#3549 (comment)](https://github.com/Klipper3d/klipper/issues/3549#issuecomment-739323618)
> 
> The ticket more or less explains what's happening and how you can solve it

Wow. Great. Will deffo give this a try. Thanks

XoBoR, 2021-12-02:
> Check this: [Klipper3d/klipper#3549 (comment)](https://github.com/Klipper3d/klipper/issues/3549#issuecomment-739323618)
> 
> The ticket more or less explains what's happening and how you can solve it

At least in my case this doesn't help (already had it in my klipper config)

> The fact that NONE of these issues have a official response or close, makes me think they really really and I mean REALLY dont care about the customer.

Couldn't agree more..

Contrasens, 2021-12-03:
> Why, always geting this... either it is on Z or E or Y, seems random. But same error... so sick of it. So much wasted print time and plastic!!
> 
> Two brand new ender 5 plus with klipper, and two new E3 turbo cards. Have tried to change driver, cable, stepper motor, added extea ground wire to psu, but nothing helps...
> 
> Running klipper Just about had enough of this now.

I'd like to help out here (I am having some problems with auto home in Marlin and wanted to switch to Klipper). I also have dual Z steppers (second one attached separately to the E1 driver, and NOT on the Z driver using a Y-cable! ). My printer is a modded CR-10 Mini.

But I cannot get the Fluidd on the Raspi to communicate with the Klipper firmware I flashed (via SD card). Can anyone point me to a step-by-step installation for klipper that works on our board?

DragonFlyer007, 2021-12-03:
> I'd like to help out here (I am having some problems with auto home in Marlin and wanted to switch to Klipper). I also have dual Z steppers (second one attached separately to the E1 driver, and NOT [...]

Post on own thread and not this. Not relevant to the topic at all..

DragonFlyer007, 2022-03-26:
So long time, no errors, untill now.

Had again random stepper errors, and then before i got that sorted, bltouch also started messing about. Changes some stepper motors, and still no go. After some day troubleshooting, it turned out to be the bltouch cable it self. Somewhere instade the cable, near, but not at the connector agains the bltouch.
Swapped the whole cable, and no more stepper errors. 

So i guess the cable sent some signals to the board witch triggered random errror messages.

Really hard to troubleshoot when you dont get a direct error message.

DragonFlyer007, 2022-05-19:
So back again.. New round off same error... Man, im really about to drop out on this for good now. So sick of it.

> Check this: [Klipper3d/klipper#3549 (comment)](https://github.com/Klipper3d/klipper/issues/3549#issuecomment-739323618)
> 
> The ticket more or less explains what's happening and how you can solve it

Btw this did not solve it for me either.

Gavalakis, 2023-09-17:
Any news? My skr mini e3 v1.2(klipper) started having the same issues and a lot of failed prints. I have noticed that my bltouch fails to probe some times but by manually pulling the probe it fixes it(dont know if it is related at all). No loose cables as far as i can tell.

DragonFlyer007, 2023-09-17:
> Any news? My skr mini e3 v1.2(klipper) started having the same issues and a lot of failed prints. I have noticed that my bltouch fails to probe some times but by manually pulling the probe it fixes it(dont know if it is related at all). No loose cables as far as i can tell.

Rhis problem can be harware related. The bltouch cables have a tendency to break without visible damage. Recommended changing the cable