bigtreetech/BIGTREETECH-SKR-E3-Turbo #64: Unable to home after install of SKR e3 Turbo in ender 3 pro

BCorbett0312, 2021-11-22:
I have a dual z axis ender 3 pro with CR Touch.  My setup was working fine before the upgrade to the new mainboard that was purchased for the dual z axis.  Whenever I try to home after the install it homes the X and Y axis but the CRTouch never works and the firmware errors out.  I have looked around and followed all the suggestions on here but have yet to find a solution that works.  

Here is the branch I have with the firmware.  
https://github.com/BCorbett0312/E3_Turbo_BLTouchNotWorking/tree/main/Marlin

https://github.com/BCorbett0312/E3_Turbo_BLTouchNotWorking/blob/main/Marlin/Configuration.h
https://github.com/BCorbett0312/E3_Turbo_BLTouchNotWorking/blob/main/Marlin/Configuration_adv.h
and
https://github.com/BCorbett0312/E3_Turbo_BLTouchNotWorking/blob/main/Marlin/src/pins/lpc1769/pins_BTT_SKR_E3_TURBO.h

In particular are the files that have been modified looking for a fix.

I even tried following this guide and moving the pins to the z stop for the CR touch
https://www.makenprint.uk/3d-printing/3d-printing-guides/3d-printer-mainboard-installation-guides/skr-e3-turbo-setup-guides/skr-e3-turbo-marlin-2-setup-part-2/#pins_BTT_SKR_E3_TURBO

But nothing.

Whenever I try to home it gives me the error about bltouch being bad and needing to send m999 to reset the machine.

Here is a pic of the wiring on the board.  I have tried it both ways with the pins being on the zprobe and vice versa. 
[image]
[image]

BCorbett0312, 2021-11-23:
I updated the firmware again and changes can be seen.  I moved the pinning back to the Z Probe spot on the board.  I am still experiencing this issue.

tome9111991, 2021-12-05:
Try with `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

BCorbett0312, 2021-12-06:
> Try with `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

I just tried this and am still having the same issue.  I am going to try a different board and see if i still have issues.  

https://github.com/BCorbett0312/E3_Turbo_BLTouchNotWorking/tree/main/Marlin

Here is the repo.  I have tried just about everything.  The only thing that is strange when following this guide

https://www.makenprint.uk/3d-printing/3d-printing-guides/3d-printer-mainboard-installation-guides/skr-e3-turbo-setup-guides/skr-e3-turbo-marlin-2-setup-part-1/#config_marlin

is the part about the endstop inverting when using BLTouch.  It says to set it to true but when that happens Auto build Marlin throws an error and says that it needs to be false.

I tried following the guide exactly as it is but it still has the exact same error.  I am able to move all the axis without homing though.  Just it fails to home on the z axis.  I am at a complete loss as to what to do.

tome9111991, 2021-12-06:
I have endstop inverting to false and works for me but i use the 5pin port for endstop.

Maybe you can swicht the wireing of your 2 pin endstop?

BCorbett0312, 2021-12-07:
Yeah let me do that real fast i will switch it back and try that and report back here

BCorbett0312, 2021-12-07:
So same thing.  I noticed that when the printer starts up the probe just sticks out the whole time...i cant even control it in the menu where it says stow or deploy.  When i had it on stock board it would go in and out without issue.  I wonder if the thing was damaged or something

BCorbett0312, 2021-12-07:
I ordered a new board and a new CR Touch.  CR touch will be here wednesday and board will be here whenever.  Hopefully i just replace the CR Touch and it works as expected.

BCorbett0312, 2021-12-10:
New CR Touch has exact same issue.  On a brand new SKR e3 turbo board.  I think there is a bug in the firmware for the CR Touch on this board.

tome9111991, 2021-12-10:
Btw.
You are use pin 1_22
Pin 1_22 is for 5pin bltouch port.

Pin 1_27
Is for zmin port

radek8, 2021-12-17:
Although I have SKR V1.4, they are quite similar with your story.
Bl-Touch works well for me.
Here are the differences from your configuration.
comment
// # define ENDSTOPPULLUP_ZMIN_PROBE
// # define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
// # define Z_MIN_PROBE_PIN P1_22 // Pin 32 is the RAMPS default

uncomment
#define USE_PROBE_FOR_Z_HOMING

Route the cable to the Z-Probe connector (pin P1.22)
Pay attention to the polarity of this cable.

KiwiTane, 2022-02-20:
Kia ora, 

any success? I have the same problem? very frustrating

BCorbett0312, 2022-02-20:
Nope... I just returned my board and swapped back to stock... Got sick of
trying to get it working

TheGunPsy, 2023-06-02:
Well, I think I fixed your issue. 
My machine was having the same fit. Z would not home after flashing the latest firmware. X and Y would home and then move to Z home position. There, it would deploy BL touch probe than just hover. Eventually failing then throwing a fit telling me to reset with M999 or whatever. 

So here is the solution I worked up. Make sure you are running Marlin 2.1x Bugfix

I am sharing my compiled firmware, however I run 2 Z ball screws and a mosquito hot end so some of my parameters are slightly different. I will include the relevant files. However, The issue was fixed when I changed the parameters in the warnings.cpp file that popped up on VS code. The Z2 axis was using the X_min_pin. I changed both of these to the BL touch port which is P1_22. Reflashed and Wallah! all is working well. 

[image]

Here are the files with my settings. 
[ConfigADVh.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/11631960/ConfigADVh.txt)
[ConfigH.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/11631961/ConfigH.txt)
[WarningsCPP.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/11631963/WarningsCPP.txt)