bigtreetech/BIGTREETECH-SKR-E3-Turbo #66: DCDC Module question

Erehr, 2021-12-10:
Can DCDC module for Mini E3 [(https://aliexpress.com/item/4000315602842.html](https://aliexpress.com/item/4000315602842.html) be used on E3 Turbo?
Checked the layout and pins/distance between them seems to match
[image]
.

radek8, 2021-12-10:
Just look in the manual

[image]

closed 2021-12-10 by Erehr

dickysoe, 2023-05-28:
Hi, Sorry, I know this has been a while and closed. but do you really need the DCDC5V on E3 turbo? 
I read it already has 5V selection on the board itself.