bigtreetech/BIGTREETECH-SKR-E3-Turbo #69: i cannot get firmware to flash from sd card!

narcoad, 2021-12-26:
I have tried to flash new firmware to motherboard and nothing happens the printer just starts as normal.  the Screen is the tft35-e3 v3.0  it says no printer attached i cannot move any axis or heat the hotend. thanks in advanced

radek8, 2021-12-26:
Do you have a problem with uploading a new FW or connecting a TFT display to a SKR board?

narcoad, 2021-12-27:
its getting the firmware to flash from the sd card

radek8, 2021-12-27:
Take an 8GB or smaller SD card.
Format to FAT32.
copy palt FW to SD card.
The name of the FW is firmware.bin
Insert the SD card into the SKR slot.
Restart the printer.
After successful upload, the file is renamed to firmware.cur
If that doesn't work, try another SD card. If that doesn't work either, use another ...

radek8, 2021-12-27:
In order for the TFT display to connect, you must have the same serial configuration speed set on both the TFT and the SKR (115200).
You must also enable serial port 0 on the SKR for touch mode.
Enable "CR10_STOCKDISPLAY" for Marlin mode.

closed 2021-12-30 by narcoad