bigtreetech/BIGTREETECH-SKR-E3-Turbo #72: BLTouch not stopping z axis

Schm1m, 2021-12-31:
Im trying to set up my (originally) Ender 3 Pro with the BTT SKR E3 Turbo with an Bltouch, but it wont stop the z axis stepper, if the probe is triggered

everything from [this](https://kay3d.com/pages/the-correct-way-to-configure-a-bltouch) guide works, until step 7) where it just doesn't stop

my configuration.h file:
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/7795002/Configuration.zip)

if any other file is required to help, just ask

radek8, 2021-12-31:
Which connector do you have the probe connected to?

Schm1m, 2021-12-31:
currently in the z-endstop connectors, so pin 1.27 i think

radek8, 2021-12-31:
comment line
// # define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
Z-axis end stop connector connect to Z-PROBE connector (P1.22)
Pay attention to polarity

Schm1m, 2021-12-31:
tried it with both polarities (as i wasnt sure), now the probe wont deploy on home-z, manual deploy over the lcd menu works

radek8, 2021-12-31:
You must first determine the polarity in the manual and connect accordingly.
Experiment-mistake system, sooner or later you will destroy something.
 
With one 3D-Touch probe, I had a problem sensing position.
I had to adjust the screw in the sensor axis.

Schm1m, 2021-12-31:
yeah, but even with the right polarity (according to the manual) the probe wont deploy on auto home

radek8, 2021-12-31:
Send Configuration_adv.h

Schm1m, 2021-12-31:
[Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/7796753/Configuration_adv.zip)

radek8, 2021-12-31:
uncomment
#define BLTOUCH_DELAY 500

Schm1m, 2021-12-31:
probe still not deploying automatically

radek8, 2021-12-31:
send video

Schm1m, 2021-12-31:
[printer.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo/files/7796793/printer.zip)

Schm1m, 2021-12-31:
as shown, the probe does not deploy, but the printer sets the heigth as 0

radek8, 2021-12-31:
Do you have the probe connected correctly?

radek8, 2021-12-31:
send photo engagement

radek8, 2021-12-31:
[image]

Schm1m, 2022-01-01:
[image]
should be the same i guess

radek8, 2022-01-01:
Check the yellow wire for poor crimp

radek8, 2022-01-01:
send a photo of the connected connector on the BL-Touch probe

Schm1m, 2022-01-01:
checked yellow wire with a multimeter, it has a negligible resistance near to 0, so it should work

[image]

radek8, 2022-01-01:
The connector from BL Touch is crooked.
The pins on the connector are short and the yellow wire is lifted.
The yellow wire may not be in contact with the connector.

[image]

radek8, 2022-01-01:
Is the connector fully depressed?

radek8, 2022-01-01:
Is the connector fully depressed?

[image]

Schm1m, 2022-01-01:
everything should be plugged in properly

radek8, 2022-01-01:
Another possible problem occurred to me.
BL Touch is a servo that is controlled by a PWM signal.
Your Marlin is out of date and out of date. Recently, Marlin had a problem generating a PWM signal. If you had these versions, it could be the cause of the problem.
Wouldn't you like to update Marlin to the latest versions to rule out this possibility?

Another option in what could be a problem is a faulty BlTouch. Could you try it in another printer that has BL Touch functional?

Another possible problem would be in the faulty pin P1.23. This could be tested by remapping to another pin, for example P1.24, which should also support PWM

Schm1m, 2022-01-01:
i will try to adapt my config to the newest version. i can only try another BL Touch (Clone) on the same printer. for the last part it would be nice of you to give me a short description of how to do it.

radek8, 2022-01-01:
OK,
If the Marlin update doesn't help and the BL Touch clone doesn't work on your printer, we remap the servo pin to another connector.

Schm1m, 2022-01-01:
tried both "versions" of the BL-Touch, nothing has changed. I also tried both, the normal marlin (v 2.0.9.3) and bug-fix (2.0.x) also neither worked.

i don't know if this is the issue, but for me it seems, the printer "forgets" to deploy the probe when homing, so the probe in the "retracted" state returns the stopping signal, as it would hit the bed, before it can deploy.

radek8, 2022-01-01:
If you let the probe eject from the LCD menu, will the probe eject?

Schm1m, 2022-01-01:
yes it ejects and retracts fine manually

radek8, 2022-01-01:
You can try to allow

#define NUM_SERVOS 1

Schm1m, 2022-01-01:
uncommented "NUM_SERVOS" and set it to 1, nothing changed

radek8, 2022-01-01:
Then I can't think of anything ...

Schm1m, 2022-01-01:
well... i got it working now

you probably fixed my initial problem with any of the changes you told me to do, but opened another one. 
by not using the option Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, "USE_PROBE_FOR_Z_HOMING" wasn't automatically called, as the comment a few lines above states. because of this i uncommented "USE_PROBE_FOR_Z_HOMING" and now it deploys AND stops after hitting someting.

so, thanks for the help and quick answers

closed 2022-01-01 by Schm1m

radek8, 2022-01-01:
But, in the configuration you submitted, "#define USE_PROBE_FOR_Z_HOMING" has been enabled.

radek8, 2022-01-01:
Mainly that it works.
Happy press.

Schm1m, 2022-01-01:
yeah, i dont know why it changed, but now im glad it works, so now onto the finetuning