bigtreetech/BIGTREETECH-SKR-E3-Turbo #73: I think I blew the Pin P2_02

yael-lorenzo, 2022-01-22:
Hello folks,

I connected a bucket converter to this pin with the reversed polarity. 
🤦 
I blew the converter and the pin stopped working.

This is the board schema
[image]

Any experience repairing it? 

Now I am wondering in which pin I can connect my chamber fan 😞

radek8, 2022-01-22:
I'm estimating it on the burned Mosfet Q4 (PL4009)
If the pin on the processor is not damaged, it will be enough to replace the Mosfet.
If it is destroyed, you can use another pin that supports PWM.
But you must not connect it directly to the fan, but via a switching Mosfet.
I use this:

https://dratek.cz/arduino/1754-pwm-15a-400w-mosfet.html?gclid=CjwKCAiA866PBhAYEiwANkIneOt9nKz1glTLhHiZ80dZ4OdXuP1WBJ37Px8V5Yxd4D0k206K_JTpLBoCOREQAvD_BwE

After a small modification, it can switch on two independent fans :-)

yael-lorenzo, 2022-01-23:
Hey Radek, thanks for the response. 
I am estimating it is the Mosffet also. The board has a LED that apparently is linked to the pin. I guess the Processor pin is ok because the LED is on or off depending the temperature.
However from the schema, the LED is after the mosfet.

Thanks for the links, I didn't even know they existed!! 
Thanks again!

radek8, 2022-01-23:
Why did you connect a bucket converter to the fan output?
If you want to connect a fan there that has a different voltage, it will only work in the on-off mode. Speed control will not work.

yael-lorenzo, 2022-01-24:
I have the chamber fan (the one that cools the mainboard) connected there. I have configured it to be `ON` at 30*Celcius. Honestly I didn't see it working with variable speed, it is `ON` or `OFF`.
When it is `ON`, it is the only fan I can hear. So I decided to place a larger fan and blow outside, not toward the components.

I completely forgot about the PWM capability on that output :\ 

I will check with your schema if the (-) is still ok after my shetty connection.

radek8, 2022-01-24:
I don't use a chamber fan, so I didn't find out if it could be regulated.
I use a control board fan. He can be regulated.

If you do not need speed control, you do not have to connect according to my borázek, but directly to the FAN output

radek8, 2022-01-24:
The SKR board regulates and switches the negative pole on the Fan, Heat and BED connectors.
If Mosfet burned down, the fan will not run at all, or still, depending on the transistor damage.

yael-lorenzo, 2022-01-25:
Hello @radek8 , thanks for your replies, I understood the pin management on your diagram 😄 
> If you do not need speed control
It was a nice to have, now let's see.

I just tested what you mentioned, I have no control whatsoever on the pin, it is like always `ON`, according to the LED that belongs that flow (diagram in my first message). I compared with the other LED that belongs to FAN0 with speed control and it brightness changes with PWM. 

I also found another thread where they speak about this mosfet, https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/238

I didn't find it before. So I might try to change the mosfet.

radek8, 2022-01-25:
It was clear from the beginning that Mofset was defective.
It could also be a faulty pin on the CPU but it's less likely.
The article you are referring to shows that this is a common problem. :-(
Some fans probably generate tips that can damage it.
I would recommend replacing the Mofset with a type that will be at a higher voltage, or connect a capacitor (according to the advice in the article) that will protect it.