bigtreetech/BIGTREETECH-SKR-E3-Turbo #75: E3 Turbo DIY pins

mlaci001, 2022-01-26:
Hi,
The description includes:

"Add one I2C bus expansion interface and one UART expansion interface for customer DIY use".

What are these PINs? Where can I find these.
Thanks.

radek8, 2022-01-26:
You can find them in the AUX-2 connector

radek8, 2022-01-26:
SDA2 -> TX2 - P0.10
SCL2  -> RX2 - P0.11

mlaci001, 2022-01-26:
Thanks a lot !
Is this motherboard suitable for Ender-3 PRO?
The hot bed (24V / 220W / 2.6 ohm / 9.2A).

radek8, 2022-01-26:
The board is named E3 as Ender3.
The screw holes are spaced the same as Ender3

The title page reads:

BIGTREETECH-SKR-E3-DIP
V1.0 a V1.1
Základní desku BIGTREETECH SKR E3-DIP uvedl na trh tým 3D tisku BIGTREE Technology CO. LTD v Shenzhenu ( http://www.bigtree-tech.com/ ). Tato deska je speciálně upravena jako přímý upgrade pro 3D tiskárnu Creality Ender-3/Ender-3 Pro, dokonalá náhrada za původní základní desku tiskárny.

radek8, 2022-01-27:
You can also use UATR0
But then you will have nowhere to connect the touch screen

[image]

radek8, 2022-02-04:
> Hi, Can I use "#define SERIAL_PORT_2 3" in Marlin?

The index for the letters RT and TX indicates the serial port number that you must define in Marlin.
RX2, TX2 set port 2 in Marlin
RX0, TX0 set port 0 in marlin

PS: I am not a representative of the manufacturer :-)

closed 2022-02-04 by mlaci001