bigtreetech/BIGTREETECH-SKR-E3-Turbo #76: TMC connection error all LOW

mlaci001, 2022-02-03:
The new E3-TURBO motherboard has the following errors:
Recv: Testing X connection ... Error: All LOW
Recv: Testing Y connection ... Error: All LOW
Recv: Testing Z connection ... Error: All LOW
Recv: Testing E connection ... Error: All LOW
The original firmware, downloaded from Bigtreetech and compiled by myself, does the same.
Please help me with what could be the mistake.
I tried TMC2209 and TMC2209_STANDALONE mode as well.

radek8, 2022-02-04:
You send the Marlin configuration files you are currently using. (not the ones you downloaded or tried somewhere a week ago)
What you used in the last compilation.

Then send a photo of the TMC2209 driver from the side of the chip,
and a photo of the SKR board without TMC2209 drivers
Thank you

mlaci001, 2022-02-04:
Thanks, it's resolved.
 I tried it on the desk without an external power supply, only with USB. When I gave external power, it started working :)

radek8, 2022-02-04:
Yes, drivers without power do not work.
Please close the ticket